nefarius/ScpToolkit #91: Had a different issue and no response. Welp, here we go again.

Narkah, 2015-12-18:
Well. The controller appears to have stopped working entirely, I must have botched the install somehow. I tested it on games that were previously working. They aren't anymore. Hmmph. Here's the new debug. Afterglow PS3 controller. 
[ScpToolkit_Log-Package_Narkah_1450082972.zip](https://github.com/nefarius/ScpServer/files/66430/ScpToolkit_Log-Package_Narkah_1450082972.zip)

Jan200101, 2015-12-18:
Have you restarted ?

CriusMacOldenly, 2015-12-19:
I noticed in setupapi.dev.log, an awful lot of things like:
([strings])
     sig:                     {_VERIFY_FILE_SIGNATURE} 09:57:46.532
     sig:                          Key      = ds3controller_e82a9329-0ea2-4ffd-b735-8bf2b5693c46.inf
     sig:                          FilePath = C:\windows\System32\DriverStore\Temp\{73488e49-a91d-336d-ac89-792169de0654}\ds3controller_e82a9329-0ea2-4ffd-b735-8bf2b5693c46.inf
     sig:                          Catalog  = C:\windows\System32\DriverStore\Temp\{73488e49-a91d-336d-ac89-792169de0654}\Ds3Controller_e82a9329-0ea2-4ffd-b735-8bf2b5693c46.cat
!    sig:                          Verifying file against specific (valid) catalog failed! (0x800b0109)
!    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 09:57:46.557
     sig:                     {_VERIFY_FILE_SIGNATURE} 09:57:46.558

All of the entries that have to do with the PS3 controller have the untrusted termination problem. It sounds to me like you do not allow untrusted drivers to be installed on your computer.

There are many websites that can tell you how to allow your windows computer to install an unsigned driver - one example is http://www.howtogeek.com/167723/how-to-disable-driver-signature-verification-on-64-bit-windows-8.1-so-that-you-can-install-unsigned-drivers/ . Feel free to google the unsigned driver issue, and find a website to explain how to do it on your OS version.

Jan200101, 2015-12-19:
I dont allow Untrusted Drivers aswell but everything (except the Windows Service) works fine

CriusMacOldenly, 2015-12-19:
From the error I think I am reading, this is an untrusted driver issue, and the driver is not being allowed to be installed on this computer because it is unsigned.
How you got yours to work is great, and if you recall how, it might work for Narkah as well. A reboot won't fix an unsigned driver issue however - he can reboot a dozen times, and this still won't work.
If you have a different solution, please post it. If you know how you got yours to work, let us know as well.

Jan200101, 2015-12-19:
I just installed it ...

CriusMacOldenly, 2015-12-19:
I tried to look at your issue as well, but it is in a foreign language, and I can not read it. Since it is a picture, I can't even google translate it. It does not seem to be helpful to this issue, so I did not bother manually typing it in to google translate - and this would not have helped me in debugging your issue anyways since google translations are not 100% accurate.

Your response of "I just installed it ..." is not detailed enough to aid me in trying to fix this issue. How certain are you that you do not allow untrusted drivers? Can you check to see?

Jan200101, 2015-12-19:
I just checked i do not allow untrusted driver and i am now going to install the newest version of SCPToolKit.
By the way the language in the Pictures there are German.

CriusMacOldenly, 2015-12-19:
Yes, I recognized the word Das from the movie Das Boot. I recognized enough of the words to guess it was german, but I don't understand german enough to even guess at your problem.
It looks like nefarius is handling your issue, so it is better that I don't do a thing there.

When I googled the specific error number that was received by Narkah, and the web page most useful appears to be: https://technet.microsoft.com/en-us/library/dd441378%28office.13%29.aspx . It really still looks like a trust issue, and the second interesting link appears to be http://stackoverflow.com/questions/23812471/installing-appx-without-trusted-certificate . However, these are merely my guesses and googles, and we can only wait and see if anything helped Narkah.

Narkah, 2015-12-23:
Didn't work. I uninstalled, allowed drivers without trusted certificates, and still doesn't work. New debug, yay.
[ScpToolkit_Log-Package_Narkah_1450852715.zip](https://github.com/nefarius/ScpServer/files/70508/ScpToolkit_Log-Package_Narkah_1450852715.zip)

CriusMacOldenly, 2015-12-23:
The debug log is filled with the exact same errors - trying to install unsigned drivers.
https://technet.microsoft.com/en-us/library/ff710454%28v=ws.10%29.aspx
Are you sure you are allowing unsigned driver installation?
Since you use windows 7, I have found info on how windows 7 users need to allow unsigned drivers:
http://superuser.com/questions/124112/use-an-unsigned-driver-in-windows-7-x64
http://ccm.net/faq/3914-windows-7-how-to-install-unsigned-drivers
and if you are willing to read Microsoft's own web pages about this:
https://msdn.microsoft.com/en-us/library/windows/hardware/dn653569%28v=vs.85%29.aspx
But I can't understand what they are saying myself. Too hard to read.
Did you use the bcdedit to change your boot configuration?

Narkah, 2015-12-24:
I did use the BCEdit, I'm restarting (again) and reinstalling.

CriusMacOldenly, 2015-12-29:
So... Did it work?

Narkah, 2015-12-29:
Didn't work. Reinstalled 3 times, controller says it is using libusBK, I believe. Driver installer pretty much just said the controller is already using the right driver, from what I could understand. Can't get a new debug right now, sorry.

CriusMacOldenly, 2015-12-30:
Did you try updating/installing the driver using the device manager?

On http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller?page=146 in the first post, it looks like that's how Morphean got his to work.

Narkah, 2015-12-30:
I'd like to know what driver to update/install. Again, it's the AFTERSHOCK
DS3, not a normal DS3. I would consider trying it with a normal DS3 if I
had a spare dongle to screw around with, but I don't. | Some driver things:
Driver Provider: libusbK Driver Date: 8/28/2015 Driver Version: 3.0.5.16
Digital Signature: USB\VID_0E6F&PID_0114 (libwdi autogenerated)

CriusMacOldenly, 2015-12-30:
Ah, I had missed you mentioning it is an afterglow ps3 controller. I don't know if that makes a difference or not. The errors I kept reading are trust errors. In other words, the drivers never even got successfully installed, and thus never get a chance to try to work. I do not know how to get past this issue since you have already told your computer to get past this issue... and it is seemingly ignoring your orders... for your own protection of course.

closed 2018-01-26 by nefarius