nefarius/ScpToolkit #93: Windows Service not starting properly

Jan200101, 2015-12-18:
Everytime I start my computer the SCP DSx Service(Ds3Service) doesnt start / crashed I dont really know what causes this but restarting it fixes it

nefarius, 2015-12-18:
Look in your Windows Event Log, in the System-Category you should find service-related errors.

Jan200101, 2015-12-18:
In the System Category ?
Cant find anything there

[image]

nefarius, 2015-12-19:
That's 'cause you filtered on the wrong source. The service itself doesn't log to the Windows Event Log. You need to filter and see what `Service Control Manager` events are there.

Jan200101, 2015-12-19:
Found it 
[image]

nefarius, 2015-12-19:
Please try if [this experimental release](../releases/tag/v1.6.208.15353-BETA) fixes the issue.

Jan200101, 2015-12-19:
It didn´t i looked in the Services and it said it was running but I cannot connect a controller and stoping it over a batch gives the responds "Service is not running"

nefarius, 2015-12-19:
The window issue is fixed but not release yet. As for your answer on the service: what?

Jan200101, 2015-12-19:
The Problem isnt fixed same issue

nefarius, 2015-12-19:
So what does the event log say now?

Jan200101, 2015-12-19:
The Picture speak for it self
[image]

nefarius, 2015-12-19:
I added some time measurements [to this release](../releases/tag/v1.6.212.15353-BETA). Please try this one and submit the log pack after reboot.

Jan200101, 2015-12-19:
[ScpToolkit_Log-Package_Jan_1450536921.zip](https://github.com/nefarius/ScpServer/files/67507/ScpToolkit_Log-Package_Jan_1450536921.zip)

nefarius, 2015-12-19:
The logs tell the service is starting within the expected time span; I'm afraid I'm out of ideas...

Jan200101, 2015-12-19:
hm... I guess I will keep using my Workaround

nefarius, 2015-12-19:
You could also try to set the service start mode from `Automatic`to `Automatic (Delayed Start)`

Jan200101, 2015-12-19:
I try that

Jan200101, 2015-12-19:
Okay it worked i am going to close this then since it isnt needed anymore

closed 2015-12-19 by Jan200101