nefarius/ScpToolkit #94: Controller moving randomly. Original Dualshock 3

Kladdkaka, 2015-12-19:
Hi :)

First of, I would like to thank you for the amazing toolkit you're maintaining :)
I got a problem though, for the last few months the controller's left analog stick have every once in a while began to move to the left-top in short pulses of 2~ seconds long that continues until I plug out and plug in the usb cable again. It have happened more and more lately. Today I tried to reinstall to see if it helped, which it sadly didn't. I don't think it's really a hardware problem, because it seems to often happen after I've alt-tabbed out of a game and opened it again, but it happens also randomly during the game. I would appreciate if you could help me :)

Thanks again for the amazing toolkit! :)

tydog98, 2015-12-19:
I have the same issue of going to the top left

Jan200101, 2015-12-21:
Only  happens sometimes and when i am playing Rocket League

Aramatheus, 2015-12-21:
I was having this same problem, controller would spaz out randomly with movement corresponding to the upper-left area of the screen. Someone somewhere (random forum post, don't remember anymore) talked about disabling "Enhance Pointer Precision" in Windows. I did that, and voilá, erratic movement stopped. Here's a link on how to turn it off, don't know if this works for you but you might as well try since it's an easy fix! [Enhance Pointer Precision](http://www.thewindowsclub.com/enhance-pointer-precision-windows)

Jan200101, 2015-12-22:
did it work for anyone else ?

Kladdkaka, 2015-12-23:
For some reason it only happens when I use my dualshock 3 with a usb cable, if I use it with bluetooth there is no problems at all.

@Aramatheus Thank you, I'll try it :)

@Jan200101 How do you make a log pack :)?

Jan200101, 2015-12-23:
@Aramatheus  To get the Log pack Search for ScpToolkit Debug Info Collector in your start menu, run it and follow its instructions.

mirhl, 2015-12-28:
Is this just a deadzones problem (aka game is stupid) or does Windows even report this big "movement"?

Jan200101, 2015-12-28:
or a faulty usb cable

njfox, 2015-12-29:
I believe I'm experiencing this issue as well on windows 8.1 with a wired ds3 controller. I tried adjusting the deadzone threshold and disabling "enhance pointer precision" but they didn't seem to make a difference.

Both analog sticks will occasionally "stick" facing the upper left part of the screen, resulting in jerky character and camera movements in that direction. Wiggling the analog sticks will usually stop the behavior for a short while.

Wehavecookies56, 2016-03-31:
I have been having this exact issue and have tried both of my DS3 controllers with multiple different USB cables and it happens every time. Here's what it looks like http://i.imgur.com/kxakp7n.gifv

DowningC, 2016-09-10:
Was this ever resolved or was a solution found? I'm on Windows 10 using an original Dualshock 3 and this problem only started when I plugged in the controller using a USB cable rather than communicating over bluetooth. When I look at the controller in windows' controller settings window, it gets sporadic pulses to the top left on the "dpad"; which sounds exactly like the behavior I'm reading about here. I also have issues with face button presses triggering multiple times even though I only press the button once (like having multiple button down events even though the button is being held).

Kladdkaka, 2016-09-11:
Sorry, I have completely forgot that I opened this ticket, it is still happening for me, but as you said @DowningC it only happens when my original Dualshock 3 is communicating over a USB cable, never over bluetooth.

The weird thing is that it seems like it's happening way more often now than before O.o

BashThem, 2016-11-21:
I'm glad i found this. Been having the exact same problems with the left stick moving to that random up-left position and wanting to start a topic.  

Although un-checking that pointer precision box in mouse settings did nothing, I'm sure this is still a mouse related issue.  When hovering over certain objects while the Controller properties window is open, I seem to get that up-left movement to occur as well.

Has anyone seemed to figure out anything else regarding this issue?  No Bluetooth here, so I am stuck trying to get this to work with USB. 😣

njfox, 2016-11-21:
I was never able to solve this issue--I ended up getting a Steam controller instead. I figured this behavior could be due to the fact that my controller was old and the analog stick was loose/worn out, but I really have no idea.

BashThem, 2016-11-22:
Well it appears it has been "solved" a while back. I'm guessing everyone having this problem has a controller MAC address starting with 08:A9:5A ? If you follow #179 and add this address to the Genuine list and restart, the problem seems to disappear. (Need to edit the .ini file in notepad as administrator to save)

I hope this helps some others. 😊

Kladdkaka, 2016-11-27:
Thank you @BashThem! :) I will try to do that, hopefully it works! :)

collincrambo, 2017-08-06:
I know this is way after the fact, but the fix is actually super simple in Windows 10. Basically it's an issue with calibration in the SIXAXIS driver. Follow these steps to fix the issue:

1. Uninstall SPCToolKit drivers in "Device Manager" and revert controller to original generic "PlayStation(R)3 Controller" driver
2. Go to Control Panel, select "Hardware and Sound" then access "Devices and Printers"
3. Right click on "PLAYSTATION(R)3 Controller" and select "Game Controller Settings"
4. Click "Properties" -> "Settings" -> "Calibrate". Follow the steps it gives
5. Go back to Device Manager and [rollback] to the SPCToolKit drivers. (If you can't find them, they should be in "C:\Program Files\Nefarius Software Solutions". Just search manually for drivers and select this folder).

6. Done. This should've calibrated everything and removed the random button-pushing issue.

mikegonzalez2k, 2017-11-01:
This has to be a Windows driver communication issue.  I have a DS3 controller from my PS3 that works perfectly still.  I tried it on my Android phone, and tablet in emulators and it NEVER does the random movement.  However when I run it ni Windows 8.1 with SCPToolkit it does random sporatic movement until I have to unplug it.  It's not calibration because that would mean it would always do this.  The fact that the controller is working perfectly one moment and the next tells me it is somehow a corruption during initialization.  Some event trigger is corrupting the data sent to Windows and Windows is receiving bad inputs from the software translation side.

mikegonzalez2k, 2017-11-01:
The problem seems to be with a change made after ScpServer v1.5.5724.38523.
The bug does exist in versions prior to this according to this thread.

https://github.com/nefarius/ScpToolkit/issues/165

closed 2018-01-26 by nefarius