nefarius/ScpToolkit #95: ScpService/server: high cpu usage even when idle

JamEnergy, 2015-12-19:
Happens on 2 machines.
Windows 10 or Windows 7.

It doesn't happen on my other Windows 7 machine though.

What are the common causes of this? 

Would any logs be helpful?

I uninstalled and reinstalled a few times.

nefarius, 2015-12-19:
What is "high" CPU usage? What are the specs of the machine(s)?

JamEnergy, 2015-12-19:
Hi there,
I should have been more specific but here goes:
The "good" machine is indeed quite powerful, like 4.2GHz. And I'm seeing SCPService consume under 1% CPU there.

The "bad" machine I'm on now is a Surface pro 3 (ok, I know they aren't power houses but it's an i7 that can boost up to like 3GHz).
It seems to be eating up 25+% CPU (I'm guessing that's one core at max + a bit).
Of course I expect this machine to exhibit higher CPU usage, but not so much more. 
[image]

[image]

The other "bad" machine is like a 3GHz i5. Again, it was using one core at max.

nefarius, 2015-12-19:
What version of the software are you using? Are you using a Bluetooth adapter? Are any controllers connected? Does this also happen if you manually restart the service from the service console?

JamEnergy, 2015-12-19:
Thanks for quick replies, and sorry for omitting such obvious information XD
Version of ScpServer and ScpService (I tried both).
1.6.126.15319

[image]

Yes, I tried stop/start from Services.msc.

No bluetooth adapter. No controllers are currently connected.

nefarius, 2015-12-19:
The release you're using if over a month old. Please update and try again.

JamEnergy, 2015-12-19:
Okay, thanks. I will do that, test, and get back to you.

Do I need to do anything special to upgrade? Or just stop services then run new installer and follow options?

nefarius, 2015-12-19:
You don't have to stop the service by your own, the installer will take care of everything and prompt you for any additional actions needed. I recommend you to [get the latest beta](../releases/tag/v1.6.212.15353-BETA).

JamEnergy, 2015-12-19:
1.6.206.15349 exhibits same behaviour.
I can try the beta too.

JamEnergy, 2015-12-19:
Same symptom in beta.

Could try ScpCleanWipe.exe (haven't yet).

Are there symbols in this build btw? Could VerySleepy it.

nefarius, 2015-12-19:
You can select installing them in the installer under `Testing & Troubleshooting`

JamEnergy, 2015-12-19:
Doesn't mean a lot to me lol, but fwiw:
[image]

nefarius, 2015-12-19:
You could do a `Save as...` and submit the `.sleepy` file so I can open and analyze it myself.

JamEnergy, 2015-12-19:
Ok, sure thing.

JamEnergy, 2015-12-19:
Google drive link to 10s of profiling on the suspect thread:

https://drive.google.com/file/d/0B15-n_tt44MeOEc3N241MzlJUUU/view?usp=sharing

nefarius, 2015-12-19:
Great, thanks. I'll have a look.

nefarius, 2015-12-19:
Can I also have you log package please.

JamEnergy, 2015-12-19:
[Logs.zip](https://github.com/nefarius/ScpServer/files/67542/Logs.zip)
Logs folder from C:\Program Files\Nefarius Software Solutions\ScpToolkit

Sorry for late reply, here's the log dir. If you meant something else, I can go and fetch that.

Thanks

Jan200101, 2015-12-19:
To get the Log pack Search for  ScpToolkit Debug Info Collector  in your start menu, run it and follow its instructions.

JamEnergy, 2015-12-19:
Oh, thanks for that tip.
Here it is.
[ScpToolkit_Log-Package_jeremy_1450551150.zip](https://github.com/nefarius/ScpServer/files/67571/ScpToolkit_Log-Package_jeremy_1450551150.zip)

Jan200101, 2015-12-22:
Question: There is a arrow next to SCPService in your Taskmanager could you send a picture of that too ? It looks like if there is more then one open

alex-mazzariol, 2015-12-31:
I'm experiencing this problem too - 1 out of 4 cpu cores is at 100% utilization by ScpService. Attaching VerySleepy log of 10 seconds and log package, may help in differential analysis.

I'm using the latest stable, 1.6.229.15365. Windows 7 x64 fully updated. The abnormal CPU utilization is happening both with or without any controllers attached, either playing or without any inputs from the controller.

[DebugInformation_20151231.zip](https://github.com/nefarius/ScpServer/files/75606/DebugInformation_20151231.zip)

brianj64, 2016-01-03:
Same issue here
[image]

Windows 10 Pro x64, freshly installed(this includes Threshold 2 aka version 1511)

All I was doing was watching some anime in Media Player Classic Home Cinema x64

ArtStepanyuk, 2016-01-12:
Im having the same issue Windows 10 intel core i5 -4430 cpu 3.00GHz 25-30% load at idle state

shoebaker, 2016-01-20:
Same issue windows 7 x64 intel at 25% idle. With controller connected or not, its always maxed one core out.
PS3 controller with cable, no bluetooth.
It all still works.

tried versions 1.6.229.15365 and 1.6.238.16010 same thing.

[ScpDsxService_asafs.log.xml.txt](https://github.com/nefarius/ScpToolkit/files/97278/ScpDsxService_asafs.log.xml.txt)

Jan200101, 2016-01-20:
My last guess would be a Virtual Controller

shoebaker, 2016-01-21:
Tried v1.5.5787.40106 and this problem does not occur. 
(previously tried v1.6.229.15365 and v1.6.238.16010)

Azzurite, 2016-02-06:
Same problem, 2/8 logical cores are fully used all the time.

I attached my ScpService log, a .sleepy analysis with debug symbols installed and my msinfo32.

[2016-02-06_issue95_high_cpu_usage_Azzurite.zip](https://github.com/nefarius/ScpToolkit/files/120200/2016-02-06_issue95_high_cpu_usage_Azzurite.zip)

nefarius, 2016-02-06:
I now know what the problem is but not yet how to fix it. Stay tuned.

JamEnergy, 2016-02-08:
Thanks for the update.
Nice for us users to see that glimmer of hope.

Good luck.

nefarius, 2016-04-02:
I think the `libusbK.sys` driver is to blame here (which I ditched in the upcoming release) so please check out the very next release to see if this fixes your issues.

NPException, 2016-12-30:
I just wanted to let you know that I had the same problem. I now installed the latest Beta, and the problem seems to be gone.

Thank you very much! :)

closed 2016-12-30 by nefarius

danyboy666, 2017-01-03:
This fix is working perfectly. Processor usage dropped from 20% to 0.1%. Thanks.