nefarius/ScpToolkit #96: Fake Dualshock 3 showing up as Genuine

yadav1aryan, 2015-12-20:
SO i have both, supported controller and Supported Bluetooth 
Controller
USB\VID_054C&PID_0268&REV_0100
and bluetooth
USB\VID_0A12&PID_0001&REV_0134
USB\VID_0A12&PID_0001

It Works Flawlessly over USB but after disconnecting, when it tries to connect via bluetooth it gets recognised as a genuine dualshock3 and errors and warnings start displaying.
Heres the log

http://pastebin.com/1Jq9et5s

Thanks, and btw this controller works with the android sixaxis app, properly.

CookiePLMonster, 2015-12-20:
Oddly enough, I have the opposite - my genuine (shipped together with PlayStation 3 Super Slim) controller shows up as fake.

tydog98, 2015-12-21:
As with CookiePLMonster, my genuine controller also shows up as fake

yadav1aryan, 2015-12-22:
you guys should try editing the .ini, you might be in luck

closed 2018-01-26 by nefarius