nefarius/ScpToolkit #99: Dpad random inputs

jross86, 2015-12-21:
While playing, every once in a while I get a short flurry of spurious Dpad inputs.  Using original ds3.

Possibly related to this issue?
https://github.com/nefarius/ScpServer/issues/94

Jan200101, 2015-12-22:
Have you tried the newest solution in #94 ?

jross86, 2015-12-22:
Enhanced Pointer precision was not checked, so that can't be the problem.  The link to 94 was speculative, starting to think what I'm experiencing is unrelated.

Jan200101, 2015-12-23:
okay a log would be nice

jross86, 2015-12-23:
Here it is:
[ScpToolkit_Log-Package_jross_1450907883.zip](https://github.com/nefarius/ScpServer/files/71301/ScpToolkit_Log-Package_jross_1450907883.zip)

jross86, 2015-12-29:
Update: problem has fixed itself.  I didn't change anything, no idea what happened.

closed 2016-04-02 by nefarius