nefarius/ScpToolkit #100: Controller works - Keyboard acts like controller when controller is disconnected

INsynth, 2015-12-22:
Controller works flawlessly so far, but when I use the keyboard and mouse, I spin really fast in different games or the mouse just flies to parts of the screen. I even uninstalled SCPtookit and the keyboard I still notice the same thing happening.

Jan200101, 2015-12-22:
This might not be be SCPServers fault maybe you (or a friend etc.) changed some setting

INsynth, 2015-12-22:
Okay, I reinstalled SCP and uninstalled it again. All of a sudden, my keyboard magically works now.
No clue what happened, but hey.

INsynth, 2015-12-28:
Welp, it's back. Here's some context:
After I uninstalled, the issue came back. Then the keyboard turned off randomly and everything was back to normal. I plugged in my PS3 controller for a different game and when I was finished playing, I started spinning uncontrollably, but only once I press a button on the keyboard and move the mouse. I'll reinstall SCP and post logs.

INsynth, 2015-12-28:
Logs:
http://www.mediafire.com/download/v2c63yie1810i1n/ScpToolkit_Log-Package_1451303339.zip

Jan200101, 2015-12-28:
Okay i havent read the logs yet (and dont understand all of it yet) but my guess is that you maybe hold the joy stick while unplugging / the driver failed to translate the ps3 controller code to xbox360 controller code and  put it to position 0,0

INsynth, 2015-12-29:
Hmm might be since the joystick of my controller is iffy at times. Well it's back to normal again, but I'll try to update if anything weird happens. Part of me thinks that I accidentally installed the driver on the keyboard, so I'll just hook-up another usb keyboard and uninstall from there.

Jan200101, 2015-12-29:
okay

closed 2016-01-09 by nefarius