nefarius/ScpToolkit #102: Original Controller not working (DS3)

xeo-it, 2015-12-23:
I've a curious couple of issues on my Win10 machine.
I've an original DS3 controller and a fake one.
Using the OLD SCPdriver (before you took over), the original controller works ONLY in BT mode, when I plug it via usb it stop working (it's detected but nothing works).
The fake one works only via usb (but that's ok since the mac address is not recognized).

With your version of scp driver, the original doesn't work at all, either bluetooth or USB while the fake one works in USB mode like before.

I've uninstalled the old drivers and tried various combinations (force, non-force) but with no luck.

Any hint? 
Thanks

Jan200101, 2015-12-26:
okay does it happen after you restart/ are there any errors while installing ?
A log pack would be nice

xeo-it, 2015-12-26:
Ok I've managed to get it working.
Basically I've uninstalled the old, restarted, installed your version, restarted again, turned on the ds3 and it connected automatically. 
After that also the bluetooth started working.
When I plug the USB cable I see a couple of errors every time (null reference exception) as a notification.
I've attached the logs if you wanna have a look.
Thanks a lot.
[scp.zip](https://github.com/nefarius/ScpServer/files/72527/scp.zip)

Jan200101, 2015-12-28:
I wont if it worked.... and thats the wrong log.... so yeah...

closed 2016-01-03 by nefarius