nefarius/ScpToolkit #103: DS3 Controller doesn't work. At a loss why not. (win10)

Potatomasher5, 2015-12-24:
I have exhausted all reasonable options: latest beta version, installing with unsigned drivers allowed, force/non force, etc. 

my situation:
-win10
-original DS3
-supported bluetooth device
-Logs included: 
[ScpToolkit_Log-Package_Arne_1450915916.zip](https://github.com/nefarius/ScpServer/files/71348/ScpToolkit_Log-Package_Arne_1450915916.zip)

-When installing drivers, I often get 2 error popups. Unfortunately I haven't been able to screenshot them, and I can't seem to get them again. They said something about nullpointerexception, and not being able to access the device. The other one seemed to say the same but for the bluetooth device. Don't quote me on this, I hope this will show somehow in the logs. I made damn sure the installer had Admin privileges, but to no avail.

-The control panel recognizes the device correctly as a PS3 controller, however it does not acknowledge any inputs.
-with one of the reinstalls I got a popup when connecting the device, saying the device was connected,  followed by 2 similar errors as above. With later installs, I didn't even get a popup. If necessary I think I can recreate this through trial and error.
-MotionInjoy was installed on this system before, however I uninstalled it following this: http://emulation-general.wikia.com/wiki/MotioninJoy#Un-Installation_instructions procedure.
-The lights on the controller flash while connected, but after a while they go out.

Please help me, I'm on the verge of a breakdown, or even worse, reinstalling the dreadful motioninjoy shite.

nefarius, 2015-12-28:
Your error where logged, so thanks for the pack :) you have a pretty low-level USB error. Are you 100% sure the USB cable you used isn't in some way damaged or worn out? Is the USB port on your DualShock wiggly or in some other way noticeably loose or broken?

nefarius, 2015-12-28:
Some quick googling after the error code `0x80004005`, maybe helpful:
 * http://answers.microsoft.com/en-us/windows/forum/windows_10-update/error-0x80004005-window-10-update-failure/156adfb8-e7e1-46fa-b5c6-cbd5dcfebbb9
 * http://www.infoworld.com/article/2960528/patch-management/windows-10-patch-kb-3081424-can-crash-fail-to-install.html
 * http://answers.microsoft.com/en-us/windows/forum/windows_10-update/cumulative-update-for-windows-10-for-x64-based/792cfa6a-c08d-4ddb-be99-71b440d51252

Potatomasher5, 2015-12-29:
Well the USB cable is a bit old and has had some issues, but now I installed the old SCP drivers (the ones before SCPtools) and it works perfectly (although not with bluetooth, but I dont really care). Now that it works Im a bit scared of trying again with a different cable, but Ill have a look at those links later.

closed 2018-01-26 by nefarius