nefarius/ScpToolkit #104: After restart / sleep bluetooth problems

erapkin, 2015-12-24:
After I restart my machine, or it goes to sleep bluetooth connections fail to be made.  Usb connections still work.  Using v1.6.206.15349 on windows 8.1.  If I uninstall the driver and reinstall it bluetooth connections work again.  This occurs consistently.

tracknChanges, 2015-12-24:
I can affirm this behavior, same version, but on windows 10 64bit.

1cup, 2015-12-26:
Hey, going to jump on this one.  I am also encountering this problem  Windows 8.1 64bit, same version of SCP as OP.

rimbas4, 2016-01-06:
Can confirm this happens on my laptop. Got bluetooth working twice. I'm guessing it has to do with BT radio going into sleep mode and then never waking up.

D-Zen, 2016-01-27:
I have a asus K43SJ laptop with bluetooth module, i install driver for buetooth module and ds3 gamepad, it's working well but when my laptop wake on from sleep mode, bluetooth module not wake on and not connect with ds3 gamepad, i used ScpToolkit 1.6.238.16010. How to fix it?

crogers3, 2016-01-27:
I can confirm this behavior as well. So far, the only fix that has worked for me is to use the clean wipe utility (while the controller is plugged into USB), and then re-run the driver installer to reinstall the controller drivers, the bluetooth driver, and the service itself.

closed 2018-01-26 by nefarius