nefarius/ScpToolkit #105: Driver for Xbox 360 Controller is removed after restart

DuskEyed, 2015-12-25:
Running the installer for the Dualshock3 controller, and using the controller is flawless. However, after I restart my computer (Windows 10 64bit) the device labeled "Xbox 360 Controller" that the installer creates, is gone. The scpsettings.exe and scpmonitor.exe do not function and throw errors after a restart.

If needed, I can provide the debug file created by the scpdebug program.

1cup, 2015-12-26:
This sounds like a problem with your xbox 360 driver and not scp server.  Make sure these are installed: 
https://www.microsoft.com/hardware/en-US/d/xbox-360-controller-for-windows

DuskEyed, 2015-12-26:
I don't think that is the issue, seeing as how the controller would not function at all if I did not have this driver installed. (I do)

DuskEyed, 2015-12-26:
If I run the installer again, the device shows up in the control panel, but consistently gets removed after a restart.

Jan200101, 2015-12-26:
after a restart open a command promt and write "NET HELPMSG 3521" in it and tell us the results.

DuskEyed, 2015-12-26:
"The *** service is not started." I'm assuming this means that it's an issue with it starting back up automatically?

Jan200101, 2015-12-26:
yep go into your services and search for "SCP DSx Service" and change the startup option from 

"Automatically" to "Automatically (Delayed Start)"
that should fix it

DuskEyed, 2015-12-26:
So I used the Services application to start it manually, and immediately heard it connect my controller. So, that was the issue, I'm going to set it to a delayed start like you said and see if that works.

Jan200101, 2015-12-26:
It should I did for me. The Problem is that the Service has some kind of error on startup when starting too early which kills it.
If it still wont work then I would try making batch for it and starting the batch when you want to use your  controller.

DuskEyed, 2015-12-26:
Having it delay the startup worked out, thanks a ton.

Jan200101, 2015-12-26:
no problem

closed 2015-12-28 by nefarius