nefarius/ScpToolkit #106: Fake Controller dump ( Gasia )

antisparkles99, 2015-12-25:
[Controller (XBOX 360 For Windows)_hid-report.dump.txt](https://github.com/nefarius/ScpServer/files/72169/Controller.XBOX.360.For.Windows._hid-report.dump.txt)

For adding support of fake controllers? The dump file ... sorry i am not very good at computers.. Hope you will add bluetooth support?

antisparkles99, 2015-12-27:
[image]

This and another Linux software have a very high compatibility with fake/knockoff controllers, is it possible to somehow port the wireless connectivity features from these apps to the latest ScpServer drivers?

nefarius, 2015-12-28:
What other Linux software? Unfortunately most android apps are closed source; I have tried to contact the author of the "Sixaxis Controller" app but got no response so far. I'd love to see some code working with those knock-offs...

antisparkles99, 2015-12-29:
https://github.com/mcherri/sixaxis-android

^ The source code and repository of SisAxis Controller

And the repository of the Ubuntu/Linux software is  https://github.com/supertypo/qtsixa

nefarius, 2016-01-09:
Ah, didn't know these, will have a peek, thanks :)

antisparkles99, 2016-01-25:
@nefarius The latest update finally connects the gasia controller!! But it gets connected, only the plugged in port #1 virtual bus message appears, but the controller dosent work, it blinks fr some time, and turns off again, without any input working...Is there a fix?

antisparkles99, 2016-01-25:
[image]

antisparkles99, 2016-01-25:
And the scpmonitor shows these values shortly befor the controller turns off

LobsterBuster, 2016-08-03:
The Doubleshock PS3 controller works straight up on windows 10.

Here is an image of the product :

[image]

closed 2018-01-26 by nefarius