nefarius/ScpToolkit #107: ScpToolkit Driver Installer

Ninja92, 2015-12-26:
"ScpToolkit Driver Installer" doesn't do anything. It won't load, I've tried everything. Please tell me what i might try to do.
[image]

Jan200101, 2015-12-26:
Thats New. What windows are you using ? It looks odd to me

jfchevrette, 2015-12-26:
Same here. Had to reinstall Windows 10 today and after a fresh install I installed ScpServer and the driver installed doesn't do anything.

Ninja92, 2015-12-26:
As I said, Windows 8.1

jfchevrette, 2015-12-26:
I have the try icon which only has the "Exit" option in the menu. Double-clicking it doesn't do anything either.

Starting the ScpToolkit Settings Manager returns an Unexpected Error: Initialization of SCPSettings.MainWindow threw an exception"

jfchevrette, 2015-12-26:
Rerunning ScpToolkit_Setup.exe told me I had to install a component (see screenshot). However I'm 100% sure that when installing initially, that component was marked as "Skip" by default so I chose not to change it.

Everything works fine after re-running setup and installing the missing dependency

Ninja92, 2015-12-27:
I got all of the components and nothing works for me. I don't know what to do. I succesfully installed on my laptop, but I need to use my controller on this computer.

antisparkles99, 2015-12-27:
Try uninstalling everything, then installing the latest visual c++, then installing everything again, using ScpToolkit Driver Installer , then run ScpMonitor.exe

antisparkles99, 2015-12-27:
Same happened for me too, the interface is made using C++ I guess, installing and everything worked fine here

Ninja92, 2015-12-27:
I've done what you said, all i get is this. Look at the screenshot.
[image]

I translate for you.
In Italian "Segnalazione problemi di Windows" means "Windows Error Reporting"

Ninja92, 2015-12-28:
I tried a lot of uninstalling and reinstalling, i still get the same error. What can I do?

nefarius, 2015-12-28:
When Windows Error Reporting kicks in it means the program has crashed and can't recover from it. You'll find additional info about the crash in the Windows Event Log (google). My interface is in German but you should be able to find the entries by following this screenshot:

[image]

Ninja92, 2015-12-29:
Thanks for your support, please tell me if you need translations from Italian to English. Do you need more informations?
[image]
[image]

nefarius, 2015-12-31:
@Ninja92 [please try the latest stable release](../releases/tag/v1.6.229.15365); I added some more error handling to the Driver Installer start-up. Also please Install the `ScpToolkit Debug Info Collector` component, run it after you encountered an error, and follow its instructions.

Ninja92, 2015-12-31:
[image]
[image]

nefarius, 2016-01-01:
Damn, what's happening on your machine @_@ please zip up and attach the files in the `Logs` folder here for my review.

Ninja92, 2016-01-01:
Please download from my dropbox.
https://www.dropbox.com/s/w4v20p6kxvpy2ji/Logs.zip?dl=0
Github keeps saying that this is not a .zip file >.>
(I'm really grateful for your help, Happy New Year)

nefarius, 2016-01-01:
Happy new year to you too :)

That's weird; is `C:\ScpToolkit\` your install path? Does this file `C:\ScpToolkit\libwdi\amd64\libwdi.dll` exist on your side?

Ninja92, 2016-01-01:
[image]

nefarius, 2016-01-01:
Do you have UAC disabled? Have you used this install folder with older versions too? Did you try to uninstall, fully delete the folder by hand and do a fresh install?

Ninja92, 2016-01-01:
I uninstalled and did a fresh install 2 times. Now for the UAC, do you mean this?
[image]

nefarius, 2016-01-01:
Did you **delete** the folder after uninstall? Yes, I mean this setting. I'd stick to Microsofts recommendation and set this to its default level or else it might have weird side effects with programs requiring administrator rights. From the logs you submitted I assume a file permission problem.

Ninja92, 2016-01-01:
I'll set to default level, unistall, delete the folder, reinstall, and try.

Ninja92, 2016-01-01:
Fresh install, still the same. Now I have only 1 file in the log folder.
https://www.dropbox.com/s/yvopb66c0e2jooi/ScpDsxInstaller_Ninja.log.xml.zip?dl=0

nefarius, 2016-01-01:
What path did you install it to?

Ninja92, 2016-01-01:
Oh god. Wait a second, this time i did a serious mistake.

Ninja92, 2016-01-01:
Do you want to continue this thread on skype? Only text no voice.

nefarius, 2016-01-01:
[I created a chat room on Github](https://gitter.im/nefarius/ScpServer), wanna switch over there?

Ninja92, 2016-01-01:
Nefarius i have some news, please enter in the chat again.

nefarius, 2016-01-01:
Nice! What was the problem?

Ninja92, 2016-01-01:
I don't know, i just did a reboot.

nefarius, 2016-01-01:
Well good for you! Have fun playing! :)

closed 2016-01-01 by nefarius