nefarius/ScpToolkit #108: PS(R) Ga`epad - Fake controller

GaucheRabbit, 2015-12-29:
Hey Nefarius,

First off, thanks for your work on ScpToolkit, wouldn't have a hope without it.

I went into this knowing it was a fake controller and figured the bluetooth would work with my laptop.
Now I'm here. I can use it just fine via USB but I was seeking freedom from the physical cables that bind.

Anyway, below are the specs and the debug info.
Holler if I can provide anything else that can help.

PS(R) Ga`epad
Model - CECHZCU
Barcode/UPC - 711719805225
Color - Transparent Dark Green

[ScpToolkit_Log-Package_Adrian M-P_1451362677.zip](https://github.com/nefarius/ScpServer/files/73724/ScpToolkit_Log-Package_Adrian.M-P_1451362677.zip)

mmbossoni, 2016-01-03:
So,
As Rabbit said, thank you very much for your work.
I pretty much have the same problem. I'm using a Dazz gamepad (http://www.dazz.net.br/Produto/621265/controle-dual-shock-bluetooth-sem-fio-azul) and it simply refuses to work using bluetooth. Dunno if it is a problem with my bluetooth dongle or the device.

Hope you could keep up the good work.

Below is the logs from debug tool (github is not letting me attach anything)
https://drive.google.com/file/d/0BzXlzIyJc44kTWgyLVpzNE9iaTA/view?usp=sharing

trigrab, 2016-01-06:
I have the same gamepad as GaucheRabbit.

Mine is white, has the same sticker, but another usb id:
HID\VID_045E&PID_028E&IG_00\3&2E05E610&0&0000
Maybe it is another one, but I have the same issue.

The problem is, that paring fails. When you pair it with the legacy scppair, it does save everything, but adds FF: in front. The Last two digits get lost.

Hope you can support this some time.

If you need any information, I'm glad to help.

trigrab, 2016-01-06:
I have it working, but I think it will never be merged into the code. Maybe we can find a different solution.

My solution:
open the ScpControl.ini
delete the line, which is my controllers mac
restart the service

Before this, I pressed the button on the back with a needle, no Idea if this helped or not. But I had to delete the mac from the ini to get it working after all.

trigrab, 2016-01-06:
The funniest part is, that the cointroller has different macs for usb and bluetooth. That means, that I have two reserved controllers in scp monitor.

mmbossoni, 2016-01-07:
At least your controller have a reset button. Mine doesn't. I've tried it on Linux using bluez as well, but no luck. It even returns wrong bluetooth device type (it tells bluetooth it is a keyboard, not a joystick).

GaucheRabbit, 2016-08-10:
Looked into #241 and thanks to @j0shg I got the bluetooth working on PS(R) Ga`epad (FiveStar) with his modified ScpControl.dll swap on version 1.6.238.16010.

I suppose this can be closed once it gets implemented in the next version.

closed 2018-01-26 by nefarius