nefarius/ScpToolkit #109: Controller is only working in the XBOX game hub but not anywhere else?

rosterx, 2015-12-30:
I installed everything correctly, I'm pretty sure. I start the SCPtoolkit server and it seems fine, no errors. If I press my PS button it will pull up the GameHub for xbox or windows 10 or whatever it is and I can use the controller to navigate it. But I can't move my mouse with it or anything. What do I do?

nefarius, 2015-12-30:
What do you mean by "you can't move your mouse with it"? Did you try it with a game? Or `joy.cpl`?

rosterx, 2015-12-31:
@nefarius Shouldn't I be able to move my mouse freely around my screen with the controller? I tried it with the game too but it didn't work. I'm trying to play Dark Souls on steam. The controller only works with the xbox app so far..Maybe I need to link the game to the xbox app somehow? And what is joy.cpl?

rosterx, 2015-12-31:
@nefarius Actually it is working, I just launched the game from the xbox app and it worked. Thanks for this software! Really easy install and works great :)

nefarius, 2015-12-31:
Glad to hear, have fun gaming!

closed 2015-12-31 by nefarius