nefarius/ScpToolkit #110: Fake controller works for a few seconds, after that it seems stuck.

Deltomane, 2015-12-30:
My fake controller (USB\VID_054C&PID_0268&REV_0100   BigBen BB4401 PS3PADRFLX) works for a few seconds, but immediately after all the controls seem stuck.
As soon as it gets stuck:
-The numbers on the SCP Monitor stop changing:
[image]
-Also the XInout Tester shows a stuck signal:
[image]
This is my Log file:
[ScpToolkit_Log-Package_Alex_1451485111.zip](https://github.com/nefarius/ScpServer/files/74880/ScpToolkit_Log-Package_Alex_1451485111.zip)
Could anyone help me?
Thanks

trigrab, 2016-01-06:
I have the same issue on the newest Version and could also solve it with the version Deltomane suggested. I will try different versions and controllers soon and then get back here, to report what I found out.

So far I tested every option on the newest Versions ScpSettings if it could help, but it didn't.

Deltomane, 2016-11-30:
I tried again with the latest beta drivers.
Still same problem...
Could anyone help me?
Thanks
My Log file:
[ScpToolkit_Log-Package_Alex_1480464204.zip](https://github.com/nefarius/ScpToolkit/files/620601/ScpToolkit_Log-Package_Alex_1480464204.zip)

closed 2018-01-26 by nefarius