nefarius/ScpToolkit #112: Suggestion: Expand to controllers other than Playstation

Nerdy314159265, 2015-12-31:
I have to say that I love the work being done on this program. I think it'd be amazing if it was expanded to include other controllers like the Wii, and Wii U controllers.

I understand that may not be likely to happen or even possible but I wanted to put it out there. I think it'd be amazing to have one program that could handle all the unofficial support for getting controllers working on PC properly.

Jan200101, 2015-12-31:
actually there might be some problems getting the Wii U gamepad to work. But you can already use Wii motes

Nerdy314159265, 2015-12-31:
@Jan200101 You can't use Wii-motes through this program though, it would require finding a separate one that doesn't have any viruses hidden like some less reputable programs I know of.

I agree that it might be a challenge but I think it'd be awesome to have a single unified program for all console controllers.

Jan200101, 2015-12-31:
actually i had one where you could write the code yourself e.g. wimote.uparrow = W etc.

Nerdy314159265, 2015-12-31:
@Jan200101 Last I checked GlovePie was dead.

Jan200101, 2015-12-31:
It is dead but still works with everything

closed 2018-01-26 by nefarius