nefarius/ScpToolkit #115: Can't use DualShock3 controller while Xbox 360 controller is Player 1

SakifX9, 2016-01-02:
What I am trying to do is make the DS3 controller player 2, but it won't let me. If I disconnect my Xbox 360 controller and connect my DS3 controller it lets me use it but sets my Xbox 360 controller to Player 2.

[ScpToolkit_Log-Package_$akif X9_1451738282.zip](https://github.com/nefarius/ScpServer/files/76232/ScpToolkit_Log-Package_.akif.X9_1451738282.zip)
[ScpToolkit_Log-Package_$akif X9_1451738304.zip](https://github.com/nefarius/ScpServer/files/76233/ScpToolkit_Log-Package_.akif.X9_1451738304.zip)
[ScpToolkit_Log-Package_$akif X9_1451738321.zip](https://github.com/nefarius/ScpServer/files/76234/ScpToolkit_Log-Package_.akif.X9_1451738321.zip)
[ScpToolkit_Log-Package_$akif X9_1451738336.zip](https://github.com/nefarius/ScpServer/files/76235/ScpToolkit_Log-Package_.akif.X9_1451738336.zip)

mirhl, 2016-01-02:
Are you using latest release(31th december)?
There's a checkbox in Service settings in ScpSetting exactly to avoid cases like this. 

@nefarius Is there any reason this is not enabled by default? Even better: why in the world one should keep that disabled?

SakifX9, 2016-01-03:
yeah I only have the latest release installed on my pc

mirhl, 2016-01-03:
AND
*did you try to check that setting*?

SakifX9, 2016-01-03:
yes I did

SakifX9, 2016-01-03:
it was the first thing I did

nefarius, 2016-01-03:
@mirhl: It's not enabled by default 'cause I was afraid (and got feedback in the meantime too) it might still be buggy - which it sadly is. I'm investigating ;)

SakifX9, 2016-01-12:
When I connect the PS3 controller it tells me it's connected to CP Virtual X360 Bus : #01 but when I check Device Manager it says CP Virtual X360 Bus : #02 I think that is contributing to the problem caused.

Schroedingers-Cat, 2016-03-11:
So while this bug is still there, is there a workaround for those who mix DS4 and XBOX360/One gamepads?

mirh, 2016-03-11:
> is there a workaround for those who mix DS4 and XBOX360/One gamepads

Waiting for the dev to fix

Or just use normal DS4 driver
Assuming we aren't talking about those shitty games that disables DirectInput when something XInput is connected, in which case you'd need x360ce

nefarius, 2016-03-11:
It definitely doesn't work reliably to mix "genuine" physical Xbox 360 controllers **and** DS3/4s due to a confirmed bug in the virtual bus driver. I'm currently testing a fixed build but even if I fix it I can't ship the fix yet 'cause I still don't own a code signing certificate (thanks, COMODO! >_<)

nileplumb, 2016-03-14:
@nefarius looking forward to the fixes nefarius! any way to help on the code signing certificate?

nefarius, 2016-03-14:
@nileplumb just a few minutes ago I dumped the request to COMODO, instead got a cert from Certum within only two days!!! Back on track!

nefarius, 2016-03-14:
It's working! They can finally live in harmony side-by-side:

[image]

Note that the genuine Xbox 360 Controllers are occupying slots 1 and 2 while the aftermarket DualShock 3 gets slot 3 assigned. What a great feeling :+1:

Schroedingers-Cat, 2016-03-14:
Now that is awesome! Looking forward for the release including this fix!

nefarius, 2016-03-14:
Now I'm not a narcissist by any means but this makes me a little bit proud :')

[image]

charlesmilette, 2016-03-14:
Is there going to be any special procedures to migrate from the old driver to this one?

nefarius, 2016-03-14:
@charlesmilette I will make the setup do all the heavy lifting and put any manual actions in the notes of the upcoming release. Next build will have some big improvements but will be user-friendly to upgrade.

nileplumb, 2016-03-14:
@nefarius excellent looks like I stepped into the ds3 game at just the right time (well maybe a couple days early) good luck hope to see a compiled release soon!

Ps will this still work fine for the pressure buttons in pcsx2 eg if I turn on PC I should have no Xbox pads connected (currently it thinks there's one, I assume that's Scp related) if I decide to use my ds3 pad it'll assign to pad 1 and I can play MGS3 or whatever fine and pressurized, then switch over to Mario kart for some 4 player action without issue all treated as xinput controllers?

Unplugging and reassigning pad numbers like 360 pads would?

mirh, 2016-03-14:
Pcsx2 uses the custom XInput.dll which you put inside its folder. 
There's nothing to worry with dolphin then.

SakifX9, 2016-03-23:
about when should we expect the next update to address this issue to come out? I'm pretty excited!

charlesmilette, 2016-03-23:
Will the new version require a root CA certificate?

nefarius, 2016-03-23:
@SakifX9 I have some testing left to do, also have to finish the new instructions in the driver installer... If everything goes according to plan maybe upcoming weekend or the one after that.

@charlesmilette no, this would not be possible since kernel-mode drivers have to be signed with a root CA cross-signed by Microsoft, we're doing this legit ;D

charlesmilette, 2016-03-23:
@nefarius Which means that there will still be that "libwdi" root certificate?

nefarius, 2016-03-23:
@charlesmilette the libwdi root certificates are created for the **user-mode** `WinUSB.sys` driver; its auto-generated on every machine so no one (not even myself) can accidentally leak any private keys that could be used to harm other users machines. It's not a mechanism I invented; it's how [libwdi](https://github.com/pbatard/libwdi/wiki) is designed. I'll think about improving this further.

SakifX9, 2016-04-08:
How's the progress going?

mirh, 2016-04-08:
Just wait for newer release, coming soon™

nefarius, 2016-04-10:
@SakifX9 @mirh [Done!](../releases/tag/v1.7.276.16101-BETA)

closed 2016-04-10 by nefarius

mirh, 2016-04-10:
[image]
I hate you

nefarius, 2016-04-10:
@mirh now why's that?

charlesmilette, 2016-04-10:
That's a lot of emails in a short amount of time:
[image]

nefarius, 2016-04-10:
@charlesmilette Oops, I guess :3

mirh, 2016-04-10:
Btw, I'm afraid "Every issue linked in this milestone" link points to wrong page (only open issues are showed, but those are closed)

nefarius, 2016-04-10:
@mirh gawddamit. Well, would've been too easy wouldn't it. Correct this another day.

candycinema, 2016-04-20:
@nefarius , this is an awesome build. The setup is very user-friendly. I'm trying to connect a wide variety of controllers at once to my PC, and it's been a long, frustrating quest with one problem after another. I use an XB1 controller wirelessly as my main controller, but having friends over, I use 2 DS3s and probably a DS4 and an Xbox 360 controller. As you can guess, it's tough to get all that working.

This version of SCPToolkit was an easy install and it helps with connecting and disconnecting virtual Xbox360 controllers to rearrange player order. However, I think the clean wipe utility deleted the SCPToolkit Bluetooth Pairing Utility and didn't replace it. In the "official" version of SCP, my controllers connected to the dongle super easily. Now, only the DS4 will, and it still won't work through this new SCPTookit.

1) Is this build focused on getting things to work without the bluetooth pair utility? I noticed that it is deselected in the features install. Is there a reason for that? Is it just too risky to try, or is it known to not function with this build? In the new SCPToolkit Driver Installer, my bluetooth dongle does not appear.
2) How can I roll back to the previous SCP I was using to troubleshoot between this build and the official download? Or maybe re-install this toolkit with the Bluetooth pairing checked off? It's worth noting that I don't have a dongle from the list of known successful dongles.
3) Is this build meant to ignore SCPMonitor? That application can only set DS3/4 controllers to player slots above them, but since it doesn't include attached Xbox controllers, it doesn't seem to be a good tool for anyone who uses both kinds.

mirh, 2016-04-20:
BT pairing is done by SCP service iirc. Pairing utility wasn't needed in the first place. 
This driver should work with both DS, so I see no reason for this problem to happen. I'd suggest to check device manager for right-wrong drivers. 
You can find previous versions [here](https://github.com/nefarius/ScpToolkit/releases). 

Last, I really can't understand what's the fuss with slot order *between* SCP devices (#45 #207) but given the problems in there are supposed to be fixed, if you want to change slot number (especially in a mixed environment) you should be using x360ce.

candycinema, 2016-04-20:
I've ensured that my bluetooth dongle has the right driver. I know pairing is supposed to be done in SCP Driver Installer, and it used to (like magic) with the legacy build, which is why I wanted to troubleshoot it. Is this failure of the bluetooth pairing worth reporting as a new bug/issue? I don't want to fill an old thread with new problems.

mirh, 2016-04-20:
> pairing is supposed to be done in SCP Driver Installer

No, pairing is not done at install time. It's handled by DS3/SCP service (idk right name now). 
Service as in "Windows service". You can see it in task manager too.