nefarius/ScpToolkit #116: [Q] Why new releases have no Setup Information file?

mirhl, 2016-01-02:
I mean, I always somewhat felt like the only way to make drivers stick was manually telling device manager "use this inf". 
Unfortunately the only one I can find in the new "installer only" versions is ScpVBus one. 

I see that with the installer forcing the driver for even the most noob that may have become completely not necessary. 
But is there any technical reason for their exclusion?

nefarius, 2016-01-03:
The driver `*.INF` files for Bluetooth and controllers are always created on demand, that's why you won't find any shipped with the setup. The driver itself is embedded in `libwdi.dll` and gets customized for every system it's installed on. That's how [libwdi](https://github.com/pbatard/libwdi) is designed so I stick to that.

mirhl, 2016-01-03:
Oh, thank you. 
I missed a lot of this new "revamped architecture" though. And there's nothing in the Wiki. 
Is there anywhere I could read on the reasons of the switch to libusb?

nefarius, 2016-01-03:
Not really, I know I'm far behind the documentation, I promise improvement ;)

The switch from `WinUSB` to `libusbK` was a bit of an accident I'm afraid when I was diagnosing problems with the DS4 and I didn't switch back because... reasons =P

mirhl, 2016-01-03:
Ehrm.. well, you are the boss. 
Thanks for your time!

closed 2016-01-03 by mirhl

nefarius, 2016-07-16:
Every method has it's own pros and cons:

- Using your standard signed INF/CAT/SYS configuration for shipping drivers is the easiest for the end-user since the installation can be handled by OS-included tools. Downside is that an additional Hardware ID _which might be perfectly compatible_ breaks consistency of the INF with the signature and you (well, me) have to re-sign it.
- Using additional tools like [libwdi](https://github.com/pbatard/libwdi) (the core that powers [Zadig](http://zadig.akeo.ie/)) overcomes the signature limitation by re-creating a self-signed INF for every device/installation. Downside of this is a breach of privacy since the end-users certificate store has to be modified and it also depends on libwdi to work on every target system.

So there is no clear winner. As a developer less code is better code so I'll stick with the first method.

mirh, 2016-07-16:
Ehrm.. You already had answered to this question :p
This, if any, replies to #380, for as much as it doesn't mention the mini-HID driver scenario.

nefarius, 2016-07-16:
Just felt the urge to clarify. =P