nefarius/ScpToolkit #119: Prerequisites Install detected as Malware by Trend Micro AV

Xenosdalord, 2016-01-03:
Hi gus ;)

Just for information, the prerequisites install process is detected as a malware by sevreal AV.
Here are the infos from Trend Micro.

Cheers ;)
Menace :	HEU_AEGISCS937
Source :	Menace
Fichiers affectés :	D:\Download\ScpToolkit_Setup.exe
Mesure prise :	Supprimé
Détecté par :	Scan en temps réel

nefarius, 2016-01-03:
[A quick search reveals this](http://www.advancedinstaller.com/forums/viewtopic.php?t=31047). Damn AV "solutions" =P It's a false-positive for you worried folks.

mirhl, 2016-01-04:
So [tell them](http://www.trendmicro.com/us/about-us/detection-reevaluation/).

nefarius, 2016-01-06:
I did; submission failed due to an unknown error... I won't trace this any further 'cause I feel like false detection is kinda the problem of the AV industry and not mine. If you don't trust me, compile it yourself, it's as simple as that with open source.

closed 2016-01-06 by nefarius