nefarius/ScpToolkit #122: Donation system?

jclevesque, 2016-01-04:
I never could make sense of all the previous SCP / Motioninjoy / Xinput *documentation* scattered over the internet in different forums (if we can call forum posts documentation). I want to thank you for making a cleanly documented and working github project (and all the efforts required to get to this state). Would you be inclined to setup some kind of donation system? I'm not super rich - I'm thinking of donating something around 10$, but perhaps some others will join in!

nefarius, 2016-01-04:
Thanks for considering a donation :) [I set a PayPal button up on my main forum thread here](http://forums.pcsx2.net/Thread-ScpToolkit-XInput-Wrapper-aka-ScpServer-Reloaded?pid=474175#pid474175).

jclevesque, 2016-01-04:
Thanks, done!

closed 2016-01-04 by jclevesque

nefarius, 2016-01-04:
Thank you kindly :)