nefarius/ScpToolkit #123: Wrong axis enumeration after gyroscope death.

mirhl, 2016-01-04:
To be honest I just know that the other day my cousin started to play with my DS3, and 10 minutes later it was misbehaving on his Galaxy Tab A. 
So I tried it on my computer and it was still malfunctioning ([here](https://cloud.githubusercontent.com/assets/9366725/12080119/6b9d37de-b251-11e5-95f9-a1ba74cd73f6.PNG) I wasn't pressing any button). 
When I hooked it to my PS3 though, everything was just fine. 

So I tried under linux.. And it was working fine even [there](https://cloud.githubusercontent.com/assets/9366725/12080137/c5cbb6fe-b251-11e5-8964-28a5c9629e4e.png). 
Astonished I tried another one, and I noticed **axis 26** was reported [differently](https://cloud.githubusercontent.com/assets/9366725/12090702/7a546f68-b2f1-11e5-87f7-bab64b5b62c9.PNG). 

Albeit I can't realize what the hell axis 4-7 and 20-22 are ([afaik](http://wiki.ros.org/ps3joy#PS3_Joystick_Buttons_and_Axes) there are just 19 axis), I'm fairly sure 23-25 are the accelerometer and the last one (the one magically not responding) is gyroscope. 
And maybe perhaps I guess it could have something to do with [this commit](https://lkml.org/lkml/2011/6/7/711). 

If it can help, disassemble reveled it is using a MSU_VX3_0.08 motherboard with SN84001 IC, Kionix KXSC4 accelerometer and Murata ENC-03R gyroscope. 

Thanks in advance for any possible insight.

mirhl, 2016-01-27:
@nefarius After thinking a bit over #170 ... couldn't it be you are wrongly querying axis and buttons?
I mean were you aware there are actually 27 reported axis by controller itself (check linux driver)?

And if for some reason some of the legit ones had to stop working... perhaps a "sequential" enumeration may get screwed

nefarius, 2016-02-06:
Thanks for your enthusiasm, I have a collection of links regarding code from the Linux world to look into but due to my tight schedule I just haven't had the time to read them yet. I'm sure the answers are out there somewhere ;)

mirh, 2016-12-14:
http://www.spinics.net/lists/linux-input/msg48347.html
https://patchwork.kernel.org/project/linux-input/list/?submitter=162701
https://www.winehq.org/pipermail/wine-devel/2016-March/112074.html

...
Yes, I'm lazy. I should check if this is fixed in Fireshock rather than procrastinate on mailing lists.

closed 2018-01-26 by nefarius