nefarius/ScpToolkit #124: Somehow differentiating libusbK driver name from that of the previous one

mirhl, 2016-01-04:
Every now and then I find myself in need of normal bt connectivity, so I have to reinstall windows own "Generic Bluetooth Radio" driver. 
Unfortunately when I open "update driver" window in device manager I end up with two identically named drivers. Could this be slightly renamed for easiness?

On a side and probably OT note: is there any way for Ds3 service not to keep bluetooth device always "opened" so that I don't have to stop it to replace this driver without rebooting/replugging receiver?
Or is "constant monitoring" required by Dualshock special querying/connection?

ps: I swear this is the last of my nitpicked bug report.

nefarius, 2016-01-09:
Sorry, name can't be changed since that's a `libwdi` thing. Service keeps the dongle occupied since it acts as the Bluetooth stack and has to respond to connection requests.

closed 2016-01-09 by nefarius

mirhl, 2016-01-10:
@nefarius It's not to be a pain in the ass, but libwdi dev said it's already possible to change driver name.