nefarius/ScpToolkit #126: Random stuttering in v1.6.229.15365

Hoshiyuu, 2016-01-05:
Updating from v1.6.202.15347 to v1.6.229.15365, both in Windows 8 and Windows 10 environment, on Genuine DS3, in both Wired and Wireless connection, random stuttering will occur every so often (a couple of times a minute) where the last input is held and no new input is registered for a couple of second.

Reverting to v1.6.229.15347 and the problem is gone, but reinstalling to v1.6.229.15365 and the problem occurs again.

Let me know if you need any other details.

trigrab, 2016-01-06:
I have the same problem and the same version as above fixes it.

I will try different versions and controllers. Till now I tested to change every option in the settings in the newest version, but nothing helped.

Also I tried version ScpToolkit v1.6.204.15348 which also doesn't have the problem.

trigrab, 2016-01-06:
I'm quite sure it is to find between v.1.6.212.15353-BETA and v.1.6.208.15353-BETA.
Thats 4 commits difference.

Hope we can fix this soon. The new version is so much better :D

trigrab, 2016-01-06:
I tried to get something out of the log files, but while playing any games the log does nothing. So I couldn't get any information out of it.

Also the error does sometimes happens more than once a minute, and then another time, I can play 5 minutes without any problems.
Maybe, my tests weren't that good if I consider this. Then we had to search all the other commits too.

trigrab, 2016-01-06:
After reviewing the commits, I think that there's a slight possibility, that the taskqueue is not unblocking after all.
Probably it blocks the service, so that there is no further communication possible for a random amount of time. Of course this depends on the windows scheduler.
Does the Ds3Service wait at any point of time till the taskqueue has finished?

nefarius, 2016-01-09:
[Please try this release](../releases/tag/v1.6.237.16008), I added a checkbox to the Settings manager where you can choose between sync and async processing. Please provide feedback if the default (unticked) setting helps your issue.

Hoshiyuu, 2016-01-10:
@nefarius after half a day of playtesting with ScpToolkit v1.6.237.16008 (As refered), I have yet to reproduce the issue with Default (unticked) setting. 

However, minimal testing (10mins~) with the async option enable does not seem to reproduce the issue too. (Without restarting PC, Launch config tool, tick async option, and close config tool, reconnect controller)

Also "experience" is typo'ed as experiance in the async description.

nefarius, 2016-01-10:
@Hoshiyuu ah thanks, typo fixed!

Hoshiyuu, 2016-01-10:
@nefarius Alright, I've put another 2 hours into testing Async box ticked. Inputs held and blocking further inputs for half a second or more happens seemingly random at 20s to 1 minute intervals and cause game to be nearly unplayable as before. 

Thanks for the temp fix! Unchecked works flawlessly.

nefarius, 2016-01-10:
@Hoshiyuu I'll test myself again using the old-fashioned way - by playing games ;)

May I ask what system you're using? Up-to-date CPU? I'm still a bit thrown off this makes things *worse* and not better on multi-core systems...

Hoshiyuu, 2016-01-10:
I tested the both modes in Rocket League so any input locks are very immediately noticeable, if it helps.

CPU is G3258 3.2Ghz Dual Core, no hyperthread, Anniversary Edition clocked at 4.5Ghz on Windows 10. Operating environment has stayed the same since November's Windows 10 update.

thecrazy, 2016-03-15:
I am getting this problem too.

@nefarius if you would like me to test something let me know. I have some spare time (between jobs) plus this bug annoys me so much I almost bought an xbox controller even tough I have a free one comming with my occulus rift in a month.

nileplumb, 2016-03-15:
@thecrazy Anyhting is possible, I actually just moved my dongles (haha, dongles *snigger*) beneath my TV via an extension cable, so that they should have better 'spread' over the room I noticed my keyboard and PS3 controller would occasionally not respond, or think I was still pressing a key for an extended period of time, only did it last night but so far so good.

My living room is fairly cable and radio frequency heavy too so I think it needs all the help it can get to not interfere. Having many WiFi networks in range, wireless 360 reciever, logitech 'unifying' wireless keyboard, Mayflash dolphinbar (bluetooth and IR), wireless printer, old microwaves in the next room and neighbouring house all within 10m (most within 2m probably). That said, I'm not sure of any way to easily test wireless interference, it does seem a bit of a gamble.

If you're that concerned, try eliminating as many factors as possible and seeking out some testing equipment/apps.

thecrazy, 2016-03-16:
Witch hunts tend to be a waste of time when you have to many suspects. I
think it would be more efficient to take a look at debug level logs from
scp and our Bluetooth adapter to confirm the nature of the problem.

closed 2018-01-26 by nefarius