nefarius/ScpToolkit #131: R2 not working properly using bluetooth

Jamaslab, 2016-01-06:
Hello,
I'm using a PS3 gamepad from BigBen (BB4410). Everything works perfectly except from the R2 stick using bluetooth. If I open the Windows Gamepad Manager the Z-axis is not stable, keeps changing from minimum to maximum. When I connect back the gamepad through the USB, everything comes back to normal.

Any suggestion?

Thank you!

truetablecat, 2016-01-07:
It doesn't work for me either, but it's slightly different for me.

According to joy.cpl, I've got the L trigger pressed down at all times. In Rocket League, my car keeps driving backwards, so that makes sense. The fault is not with the controller, as it works fine when cabled and on my PS3.

Nothing detects any inputs from L and R triggers, except the profile manager. Profile manager detects them just fine. I've tried both with the very latest drivers, and some considerably older. Same issue.

Zareli, 2016-01-12:
AFAIK it is associated to certain bluetooth adapters. On earlier SCP versions it didn't work at all, now everything but the Z-Axis works.

truetablecat, 2016-01-12:
I guess that makes sense, especially since I'm using a cheap Chinese Bluetooth adapter. However, as I said, the profile manager recognizes R2 and L2 correctly, which should rule the Bluetooth adapter out ... right?

Zareli, 2016-01-12:
The profiler also works correctly in my case (I opened a github issue a while ago, you can check it). Probably the profiler works when mapped as buttons, but the BT adapter can't work a Z-Axis.

truetablecat, 2016-01-12:
Yeah, that makes sense.

I guess I should just get another Bluetooth adapter, that costs more than 2 dollars.

closed 2018-01-26 by nefarius