nefarius/ScpToolkit #132: Request: Update notification

blubbsy, 2016-01-06:
Would be nice if the SCP server could check periodically (once per month? on startup?) if there is an update of the software and show then a notification in the windows notification center.

blubbsy

nefarius, 2016-01-09:
Your wish will be granted!

nefarius, 2016-01-09:
Done; check out the latest release :)

closed 2016-01-09 by nefarius

PLYoung, 2016-01-15:
How do I turn this off or even auto-updates for that matter?

blubbsy, 2016-01-18:
I think its not necessary to check for updates each time when there is an internet connection or when you turn on the machine (even from standby). Maybe once per month is enough? or at least that the popup does not show up when the version is up to date.

goblinpope, 2016-03-02:
I just discovered (and not by going through the code) that the updater is controlled with the Windows task schedule. Just go in and edit it to your liking or disable/remove it altogether.

bdefore, 2016-03-07:
@goblinpope can you elaborate? i couldn't find it in Schedule Tasks of Windows 10.

also, the updater does should be removed when uninstalling. presumably there's nothing to update?

deefo0, 2016-03-07:
@bdefore I found a task call "updater" in the main Task Scheduler Library folder (not under Microsoft or any task subfolders).