nefarius/ScpToolkit #133: Enhancement: more options to the system tray

trigrab, 2016-01-06:
Hi,

I like your new work. I'm just trying everything and had also luck in using a controller that never worked at the pc before.

But while testing everything I missed a function to simply restart the service. I wrote a batch for that, but I always have to run it as administrator and it's not comfortable like this.

Also I wanted to have a possibility to shut down a controller, while another one is still connected. If you restart the service, every controller is turned off.

It could be handy to have this on the context menu of the system tray.

In advance I would suggest the system tray to open a similar window like on the legacy monitor. There also would be nice to have buttons for the other two suggestions.

But don't get me wrong. I'm impressed by your work and just hope you like my suggestions or even wanted to implement them yourself : )

Zivnarb, 2016-01-06:
you can shut off a pad by holding L1+L2+ps button for a few seconds.

trigrab, 2016-01-06:
Oh cool, thank you :D

Zivnarb, 2016-01-07:
yw. ;)

can't really remember where I read about it, think it goes back to the original version.

nefarius, 2016-01-09:
The current tray tool is just a quick hack to display messages to the user; its functionality will be merged in a new tool (`ScpToolkit Control Panel`) which will also supersede the Monitor tool.

trigrab, 2016-01-10:
Okay, thank you very much for the information.

I will close this issue and wait for a release : )

closed 2016-01-10 by trigrab