nefarius/ScpToolkit #134: profile manager: swapping trigger/shoulder binds both to shoulder.

Zivnarb, 2016-01-06:
if you bind the buttons like this:

nefarius, 2016-01-09:
Can confirm, working on a fix.

Zivnarb, 2016-01-09:
short addendum: played shadow complex remastered with the profile active earlier today, even when the profile manager showed default key bindings (the profile was ticked active but after opening the profile manager showed the default layout - another bug maybe?), L2/R2 activated both L1 and L2 (and R1/R2 respectively), L1/R1 worked like normal; the game behaved as if I'd have bound L2/R2 to two buttons at once. after deleting the swap profile and restarting the service it worked like normal.

don't know if that's an issue with the game, unreal engine 4 for or the driver/profile manager acting weird (haven't tried it with another game yet). profile manager also didn't open the first time I installed the driver. after uninstall, clean wipe and deleting all files it worked after the second install - but then I managed to crash it twice. well, it's still unstable. ;)

nefarius, 2016-01-09:
Well, the profile manager is still highly experimental...

marcelodcs, 2016-04-03:
sooo...was this bug fixed yet?

charlesmilette, 2016-04-03:
Funny. The picture has the same background color as the GitHub Dark UserStyles:  
[image]

Anyways, even if this bug gets fixed, some other errors may be encountered because it is highly experimental, as @nefarius said. Be careful when using it.

nefarius, 2016-04-03:
@marcelodcs nope.

closed 2018-01-26 by nefarius