nefarius/ScpToolkit #135: DS4 Bluetooth problem (DS4 results as Reserved in monitor when unplugged)

lowlize, 2016-01-07:
Hi and thanks for the great work. I'm having a problem using my DS4 with bluetooth. I'm using an integrated Realtek Bluetooth 4.0 USB\VID_0BDA&PID_B721 , which is not on the list of the compatible bluetooth devices. When I open the monitor with the DS4 plugged in I see the Host Address and the Pad One DS4 charged, but when I unplug the controller the Pad One becomes Reserved and the light on the pad is off (if I press the PS button it flashes for a bit and then goes off again). Is the problem the uncompatible bluetooth device or something else? By the way it works perfectly while connected by USB cable. Thank you.

[ScpToolkit_Log-Package_lowli_1452169416.zip](https://github.com/nefarius/ScpServer/files/81214/ScpToolkit_Log-Package_lowli_1452169416.zip)

IlianBurgos, 2016-02-09:
Hi, same problem here, did you find any solution?

lowlize, 2016-02-09:
Nope, no one answered on the main topic either...

closed 2018-01-26 by nefarius