nefarius/ScpToolkit #138: What are the differences between ScpToolkit and DS4Windows?

kq01526, 2016-01-09:
Hi,

anyone here who could explain what the differences between [ScpToolkit](https://github.com/nefarius/ScpServer) and [DS4Windows](https://github.com/Jays2Kings/DS4Windows) are?

Regards

Zivnarb, 2016-01-09:
ds4windows only works for dualshock 4, scp (or rather the original version) started out as a wrapper for dualshock 3, which later included dualshock 4.

see: http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller

kq01526, 2016-01-09:
Thanks, but that doesn't really explain the difference.

I was aware that DS4Windows only works for the DS4, not for the DS3 (hence it's name).

What I really wanted to know is how they differ feature wise.

Assuming you only have a DS4 and no DS3, is there anything about ScpToolkit that would make you use it over DS4Windows or vice versa?

Or in other words:

Which is better? ScpToolkit or DS4Windows?

Zivnarb, 2016-01-09:
I never tried ds4windows or inputmanager (there are probably a few more), so I can't really say which one is better for your case. for example the original scp had a working profile manager, but doesn't work with the latest win10 version (TH2 or whatever it's called). current scp works but profiles for button layouts are still a work in progress; you can turn off pads with a button combination, dunno if other driver kits support that.

there's probably also a difference how they emulate the touchpad of the ds4, sound (if that's even possible), how they support multiple pads or pads in general (I _think_ ds4windows and inputmanager have a different approach, but I might be wrong on that) etc. 

if you just want to use 1-2 ds4 with a normal button layout they should all work pretty much like expected. ;)

kq01526, 2016-01-09:
Thanks. InputMapper is not open source, so it's not an option.

Does ScpToolkit allow to use the DS4 touchpad as a mouse?

And does ScpToolkit allow to control the DS4 lightbar?

DS4Windows allows to do that. Does ScpToolkit allow it as well?

Jorchking, 2016-10-09:
Bump! I want to know an answer for kq01526 question too.
Does with SCPToolkit alow to use DS4 touchpad as mouse?

closed 2018-01-26 by nefarius