nefarius/ScpToolkit #139: Rename repository from "ScpServer" to "ScpToolkit"?

kq01526, 2016-01-09:
Hi,

why is this repository called "ScpServer" when it is referred to as "ScpToolkit" pretty much everywhere else?

Why not just rename the repository from "ScpServer" to "ScpToolkit"?

Regards

nefarius, 2016-01-09:
Because it would pretty much break all links circulating around?

kq01526, 2016-01-09:
> @nefarius wrote:
> 
> Because it would pretty much break all links circulating around?

Not sure, but AFAICT you are wrong, see:

https://github.com/blog/1508-repository-redirects-are-here
https://help.github.com/articles/renaming-a-repository/

:wink:

nefarius, 2016-01-09:
Uuuuhh, fancy! I'll be brave and try then =]

nefarius, 2016-01-09:
Great Scott, it worked! Thanks for the suggestion and the links!

closed 2016-01-09 by nefarius