nefarius/ScpToolkit #140: Screen is filled with Exception messages when I run the driver installer

InputUsername, 2016-01-09:
I have no idea what I'm doing wrong, but every time I run the installer or whatever it is -- to be honest I think it's kind of confusing... but I digress -- my screen is spammed with exception messages, forcing me to end the tasks through task manager. I'm going to have to resort to using the original ScpServer for now, because this is simply unusable. :(

If I'm doing anything wrong, please enlighten me, I'd love to be able to use this.

kcville, 2016-01-10:
Same issue here. The error is that Dbreeze can't create a table because the file is already open by another process.

It happens right after it finishes installing. Windows floods the screen with the exception until I hit uninstall and then it goes away.

kcville, 2016-01-10:
Tried this version and it seems like it may work, at the very least I don't have the exceptions.
1.6.206.15349

phatchman, 2016-01-10:
Similar issue with repeated exception message on Windows 10. I tried to manually grant permissions to the files in question, but that does not appear to have resolved the issue. Exception message from log file below.

<log4j:event logger="ScpControl.ScpHub" timestamp="1452404351763" level="DEBUG" thread="1"><log4j:message>Root hub stopped</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="DESKTOP-JVED038" /><log4j:data name="log4japp" value="ScpServer.exe" /><log4j:data name="log4net:UserName" value="DESKTOP-JVED038\phatc" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="DESKTOP-JVED038" /></log4j:properties><log4j:locationInfo class="ScpControl.RootHub" method="Stop" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.ScpDevice" timestamp="1452404351764" level="ERROR" thread="17"><log4j:message><![CDATA[Unexpected error: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> DBreeze.Exceptions.DBreezeException: Creation of the database folder failed! ---> DBreeze.Exceptions.DBreezeException: Database is not operable, please find out the problem and restart the engine! FSR INIT FAILED: C:\Program Files\Nefarius Software Solutions\ScpToolkit\Db\_DBreezeSchema ---> System.UnauthorizedAccessException: Access to the path 'C:\Program Files\Nefarius Software Solutions\ScpToolkit\Db\_DBreezeSchema' is denied.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at DBreeze.Storage.FSR.InitFiles()
   --- End of inner exception stack trace ---
   at DBreeze.Storage.FSR.InitFiles()
   at DBreeze.Storage.StorageLayer..ctor(String fileName, TrieSettings trieSettings, DBreezeConfiguration configuration)
   at DBreeze.Scheme.OpenSchema()
   at DBreeze.DBreezeEngine.InitDb()
   --- End of inner exception stack trace ---
   at DBreeze.DBreezeEngine.InitDb()
   at ScpControl.Database.ScpDb..ctor()
   at ScpControl.Profiler.DualShockProfileManager.LoadProfiles()
   --- End of inner exception stack trace ---
   at System.RuntimeTypeHandle.CreateInstance(RuntimeType type, Boolean publicOnly, Boolean noCheck, Boolean& canBeCached, RuntimeMethodHandleInternal& ctor, Boolean& bNeedSecurityCheck)
   at System.RuntimeType.CreateInstanceSlow(Boolean publicOnly, Boolean skipCheckThis, Boolean fillCache, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, Boolean nonPublic)
   at ScpControl.ScpCore.SingletonBase`1.CreateInstanceOfT()
   at System.Lazy`1.CreateValue()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Lazy`1.get_Value()
   at ScpControl.RootHub.OnHidReportReceived(Object sender, ScpHidReport e)
   at ScpControl.ScpHub.OnHidReportReceived(Object sender, ScpHidReport e)
   at ScpControl.Usb.UsbDevice.OnHidReportReceived(ScpHidReport report)
   at ScpControl.Usb.UsbDevice.HidWorker(Object o)]]></log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="DESKTOP-JVED038" /><log4j:data name="log4japp" value="ScpServer.exe" /><log4j:data name="log4net:UserName" value="DESKTOP-JVED038\phatc" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="DESKTOP-JVED038" /></log4j:properties><log4j:locationInfo class="ScpControl.Usb.UsbDevice" method="HidWorker" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.ScpDevice" timestamp="1452404351767" level="DEBUG" thread="17"><log4j:message>-- Usb Device : HID_Worker_Thread Exiting</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="DESKTOP-JVED038" /><log4j:data name="log4japp" value="ScpServer.exe" /><log4j:data name="log4net:UserName" value="DESKTOP-JVED038\phatc" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="DESKTOP-JVED038" /></log4j:properties><log4j:locationInfo class="ScpControl.Usb.UsbDevice" method="HidWorker" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.ScpHub" timestamp="1452404633756" level="DEBUG" thread="1"><log4j:message>Initializing root hub</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="DESKTOP-JVED038" /><log4j:data name="log4japp" value="ScpServer.exe" /><log4j:data name="log4net:UserName" value="DESKTOP-JVED038\phatc" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="DESKTOP-JVED038" /></log4j:properties><log4j:locationInfo class="ScpControl.RootHub" method="Open" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.ScpHub" timestamp="1452404633772" level="DEBUG" thread="1"><log4j:message>++ C:\Program Files\Nefarius Software Solutions\ScpToolkit\ScpControl.dll 1.6.237.16008</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="DESKTOP-JVED038" /><log4j:data name="log4japp" value="ScpServer.exe" /><log4j:data name="log4net:UserName" value="DESKTOP-JVED038\phatc" /><log4j:data name="log4net:Identity" value="" / [... cut ...]
   at ScpControl.Usb.UsbDevice.HidWorker(Object o)]]></log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="DESKTOP-JVED038" /><log4j:data name="log4japp" value="ScpServer.exe" /><log4j:data name="log4net:UserName" value="DESKTOP-JVED038\phatc" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="DESKTOP-JVED038" /></log4j:properties><log4j:locationInfo class="ScpControl.Usb.UsbDevice" method="HidWorker" file="" line="0" /></log4j:event>

nefarius, 2016-01-10:
Does this occur on a fresh installation or on an update?

InputUsername, 2016-01-10:
Happens on a fresh install.

closed 2016-01-10 by InputUsername

Hankooc, 2016-11-20:
My Laptop (Sony vaio sve14117flb) has a BT/WIFI combo card and I've installed a software owner that switch turn on/off the wireless functions. When I turn on Bluetooth function and install the software It not runs by bluetooth only through USB cable. How may I get It runs by bluetooth?