nefarius/ScpToolkit #143: Left Stick sometimes inputting left on its own and Genuine DS3 being registered as false .

dj-jazzy, 2016-01-10:
(Main) The left stick sometimes has phantom inputs where it will constantly go to the left for a good 5-20 seconds. Even when I move the stick in the other direction and hold it there, the stick still has the phantom inputs.

(Minor) When i plug in my DS3 it says that my controller isn't genuine (but it was the original one i got with my ps3). I did cut the connections of the rumble inside the controller though... maybe that has something to do with it?

[ScpToolkit_Log-Package_Mangaramu_1452444148.zip](https://github.com/nefarius/ScpToolkit/files/84183/ScpToolkit_Log-Package_Mangaramu_1452444148.zip)
[ScpToolkit_Log-Package_Mangaramu_1452444253.zip](https://github.com/nefarius/ScpToolkit/files/84184/ScpToolkit_Log-Package_Mangaramu_1452444253.zip)

mirhl, 2016-01-10:
I wonder if this couldn't have something to do with how axis are enumerated #123

warmbaguettes, 2016-01-17:
I have the same problem, but the Left Stick moves upward. Tried different versions of the driver but the same thing happens...

DragonWolf5589, 2016-02-09:
I have the same issue.

I get warnings that I have a fake controller and when i was driving in a game, the truck keeps randomly steering off the road and locks up.

BOTH the left and right sticks keep going diagonal left-up for me (not psychally touching them, the games all thing i do that)

the work around for me is increase the deadzone. it seems to have stopped a lot of it, but still randomly locks out.

also when I plug it in USB it starts to charge, and get "fake" messages but it will not work at all, but unplug it, same "fake" messages again, but the controller works! very strange and annoying

closed 2018-01-26 by nefarius