nefarius/ScpToolkit #145: Pad ID is assigned after Port number on virtual bus, not actual Windows slot order

mirhl, 2016-01-10:


nefarius, 2016-02-06:
Yepp, that's a problem with the vbus driver. Hopefully I'll come up with a solution. Debugging kernel-mode drivers sucks though X_x

nefarius, 2016-04-02:
This is fixed in the next release.

closed 2016-04-10 by nefarius

SneakyLittleman, 2016-07-03:
I think I should report my issue in this thread: 
I have a DS3 controller and an XBOX one controller, both working wirelessly. I set my preferred device to the Xbox one controller in usb gamepad settings in windows. YET, the DS3 pad always get slot one and is always the actual preferred device in all games. 
I couldn't find an option in scptoolkit to change this order when the pads differ.
Any help?
Great work in any case. This is only a tiny issue :)

mirh, 2016-07-03:
No you are wrong, this was just about things using this driver and the *lights* on dualshock controllers. 
And "preferred device" setting is just something for *old DirectInput* games. 

What you are talking about is mixed (probably XInput) cases scenarios. 
If the problem happens *regardless* of connection order, bump #207. Otherwise #45.