nefarius/ScpToolkit #146: Emulated XBOX 360 controller GUIDs are per-slot, not per-original controller GUID

mirhl, 2016-01-10:


nefarius, 2016-01-10:
Uh sorry but what exactly do you mean by that?

mirhl, 2016-01-10:
Every controller has a unique identifier, right?
If I had two real x360 controllers, program would see them every time with a code like 028e045e-0000-0000-0000-504944564944 or similar. And they would be able to remember per-controller preferences. 

With SCP I noticed in x360ce I always get the same one, no matter which of mine controllers I attach. 
Now, of course they are two identical DS3, so I'm not actually caring, but I see many people with "mixed vendors" compatible controllers would appreciate this.

nefarius, 2016-03-12:
Is this the value you're talking about?

[image]

charlesmilette, 2016-03-12:
I find it a bit akward to have like 4 entries in the device manager, while I plugged only 2 unique DS3 controllers in my computer. A virtual device for each real device seams more appropriate than a virtual device per player slot.

nefarius, 2016-03-12:
@charlesmilette that's a bug.

mirh, 2016-03-12:
Yes, those identifiers. They shouldn't be slot-fixed.

nefarius, 2016-03-12:
I see, I'll look into it.

nefarius, 2016-07-16:
I took a close look and it seems that the `IG` (Device instance GUID) that x360ce reports is composed of multiple sources. Once I figured out how it's composed I'm positive I can solve this. Will browse through x360ce`s sources...

closed 2018-01-26 by nefarius