nefarius/ScpToolkit #148: Integrating existing solutions for DirectInput Force Feedback forwarding

Hoshiyuu, 2016-01-10:
Rumble features currently does not work with games that sends rumble request through DirectInput only.

A Japanese developer Masahiko Morii wrote a translation library (not sure about the proper term) that forwards such rumble request. http://lavendy.net/special/driver/xi/index.html (Chrome warns malware site, but it is safe and have been there forever)

A steam user NUK3 TH3 WHAL35 furthers modifies it to add more controller ID to support misc controllers for this feature. http://steamcommunity.com/discussions/forum/1/541907867790900215/

Current usage: After installing SCP Toolkit, installing the executable provided by NUK3 TH3 WHAL35 will enable rumble feature perfectly in DirectInput games.

Would be nice to see this solution integrated into SCP Toolkit, but by no means a high priority. Thanks!

TheLastRar, 2016-01-10:
Is NUK3 TH3 WHAL35's driver open source?

Hoshiyuu, 2016-01-10:
NUK3 TH3 WHAL35 only modified the driver. I am not familiar with open source licensing details, the original driver on the Japanese site does not mention licensing details.

However the source code is openly provided by the original Japanese developer, whether that qualifies it properly open source, I have no idea.

Source file, downloaded from the linked Japanese site:
[xi.source.zip](https://github.com/nefarius/ScpToolkit/files/85210/xi.source.zip)

mirhl, 2016-01-16:
This driver purpose is making DS3 and DS4 controllers completely emulating an X360 controller. 
That hasn't this. 
*Besides*, that japanese driver has already been reported to be [bugged](http://community.pcgamingwiki.com/topic/1436-need-for-speed-underground-1-random-crashes/). 
Other solution like Xinput plus should be preferred. Similraly though, I don't see why that should be logically included in these drivers.

Squall-Leonhart, 2017-04-15:
Xinput Plus only implements constant force vibration, it doesn't have any support for periodic effects, or directional vibration.

mirh, 2017-04-15:
In other news, I just remembered https://github.com/nefarius/ScpToolkit/commit/85cc28dd30ff9489da047b82f4b8cd7c743c307c#comments
I wouldn't know if it supports FFB, if ViGEm has anything comparable, and of course without #243 it's no "universal" solution for all controllers.. But still. 

Also, if it doesn't work check http://forums.pcsx2.net/Thread-Dinput-Wrapper-for-SCP-Driver?page=4#pid485356

Squall-Leonhart, 2017-04-15:
i do hear scp now exports vibration commands, but in the case of Xoutput it doesn't work since xoutput has not implemented vibration.  I am talking a  bit with tapeq to see what could be done there.

mirh, 2017-04-16:
Oh, interesting. 
Something more on the driver level, or just like a reverse x360ce?

closed 2018-01-26 by nefarius