nefarius/ScpToolkit #149: Illegal use of function IoCreateDeviceSecure - might prevent other drivers from being installed

shauleiz, 2016-01-11:
File: ScpVBus/bus/pnp.c
Line: 570
Code:

status = IoCreateDeviceSecure(FdoData->Self->DriverObject, sizeof(PDO_DEVICE_DATA), NULL, FILE_DEVICE_BUS_EXTENDER, FILE_AUTOGENERATED_DEVICE_NAME | FILE_DEVICE_SECURE_OPEN, FALSE, &SDDL_DEVOBJ_SYS_ALL_ADM_RWX_WORLD_RWX_RES_RWX, (LPCGUID) &**GUID_DEVCLASS_X360WIRED**, &pdo);

The one-before-last parameter is GUID_DEVCLASS_X360WIRED which is the GUID for class XnaComposite  ({D61CA365-5AF4-4486-998B-9DB4734C6CA3})

This is illegal. It caused me much trouble in my own project!
Please read carefully the explanation for this parameter [here](https://msdn.microsoft.com/en-us/library/windows/hardware/ff548407(v=vs.85).aspx).

nefarius, 2016-01-20:
Thank you very much for your input; I'm currently very busy but I hope I can build a fixed version soon!

nefarius, 2016-03-14:
I fixed it! Thanks again for your contribution :+1:

closed 2016-03-14 by nefarius