nefarius/ScpToolkit #153: bluetooth not working even though the controller "connects to it"

LichBreed, 2016-01-11:
hey having problem to get the controller to work via bluetooth.
have installed the dirver while the controller and the dongle both plugged in,the scp monitor shows host address,in devices and printeres the bluetooth listed as BTH DS3 device.
when i turn the controller on wirelessly a message pops up saying plugged in port#1 on virtual bus and the monitor says bluetooth 000000 none on pad 1 but the controller still keeps blinking and gets turned off after a while.
anyone has any clue about how to fix it?
Edit:running a fake ds3 and win7 
[ScpToolkit_Log-Package_Vova_1452548164.zip](https://github.com/nefarius/ScpToolkit/files/86432/ScpToolkit_Log-Package_Vova_1452548164.zip)

Dorque, 2016-01-12:
I did a search on your bluetooth adapter VID and PID from your logs and I got some generic bluetooth 2.0 adapter here: http://amzn.com/B001EBE1LI.

I don't think this adapter supports EDR. To connect to a DS3 controller, you need at least Bluetooth 2.0 + EDR. For DS4 I believe you need at least 2.1+EDR but don't quote me on that since I don't have a DS4 yet.

gawron89, 2016-01-14:
i have exactly the same issue with ds3, and i have asus bt400 which is officially supported by scp and has EDR for sure

war1ox, 2016-01-22:
I have had this problem ever since the last 2 or 3 releases. The previous release worked flawlessly! Is there a way to download previous versions of this tool?

I install everything correctly, and even restart. The DS3 says connected to port 1, and it triggers the connection sound when powered wirelessly. The problem is that it doesn't seem to work at all in games. Not even wired.

alansublime, 2017-10-20:
gawron89 friend works without problem

TheVirusWar, 2018-01-23:
I have to reply to this topic even tho its old. I just have to.
I've been killing myself to make my controller work with bluetooth and just couldnt do it.
First I tried to do it on the Windows 10 and gave up after of literal days of tries. But now I reinstalled Windows 7 (for fun) and later I came across the PS3 controller I bought from some guy for 5$, it had a busted usb port and I've re-soldered a new one (got 15 female usb connectors off aliexpress for less than a buck!), so I decided to try again.
And as the last time, I've been trying to make it work for a couple of hours until I came across this topic and the fella up there, @war1ox, saying v1.6.133.15324 works. Had nothing to lose, gave it a try.
...
It f#$king does work. IT F#$KING WORKS!
I didnt do a damn thing! Just installed it and worked out of the box!
-btw: I have a legit sixaxis PS3 controller and a bt dongle CSR8510 A10 (VID_0A12&PID_0001); Windows 7 also (I dont think it matters, but just putting it in here)

closed 2018-01-26 by nefarius