nefarius/ScpToolkit #154: Annoying Notifications

RaTcHeT302, 2016-01-12:
Can these be disabled? They are seriously annoying.
http://i.imgur.com/bYANomp.png

Jan200101, 2016-01-13:
No and Yes.
You can change the sound of it but I dont know if you can hide it yet

Hoshiyuu, 2016-01-13:
For now you can either untick SCPTrayApp or remove SCPTrayApp.exe from installation folder and it'll work just fine.

You can disable the sound individually or fully by going to the ScpToolkit Settings Manager.
[image]

mirhl, 2016-01-16:
Delete ScpToolkit Tray Notifications shortcut from `%ALLUSERSPROFILE%\Microsoft\Windows\Start Menu\Programs\Startup`

nefarius, 2016-02-06:
I will add some more settings to allow more control over them in a future release.

closed 2016-04-10 by nefarius