nefarius/ScpToolkit #155: Adaptor VID_0E8F&PID_0003 not working

guilegatan, 2016-01-12:
Hi!
I have an USB-adaptor for DualShock 2 pads, which is recognized as "PS(R) Gamepad Adaptor". It has the hardware-id USB\VID_0E8F&PID_0003. A Google-search says the same chip is used in multiple 3rd party wired PS3 pads.

It installs fine at first, and ScpToolkit identifies it as "Genuine Sony DualShock 3 detected".
[image]

But it recognizes no button presses. Also the led-indicator on the adaptor is off. Pressing the "Analog"-Button does nothing. On real PS3, the "Analog"-Button is useable as Home-button. After pressing it, the led-indicator shows the assigned controller number.

Any chance that this adaptor will be supported anytime?

thanks,
keep up the great work!

mirhl, 2016-01-16:
Attaching debug information like mentioned in the forum thread wouldn't hurt in the first place.

guilegatan, 2016-01-16:
Sorry .. yes .. you are right.
Here is the debug info!
[ScpToolkit_Log-Package_Martin_1452951469.zip](https://github.com/nefarius/ScpToolkit/files/92899/ScpToolkit_Log-Package_Martin_1452951469.zip)

Mugi4ok, 2016-01-24:
Mine DS2 adaptor (from DreamGEAR) has the same ID, behavior is same, recognized as "USB Joystick".

mirh, 2016-07-16:
I guess #243 ?

closed 2018-01-26 by nefarius