nefarius/ScpToolkit #158: No Controller in "Game Controller" panel

ZeekDaGeek, 2016-01-14:
From everything that I can tell everything has installed properly, my controller has paired and connects without issues, however, no controller is created in the "Game Controller" panel and as a result it doesn't seem that I can use the controller in any of the game that would otherwise accept XBox controllers.

I've used an official Sony DualShock 3 using the latest version of SCP available through this github.

When originally plugged in prior to installing a Playstation controller appeared in the controller panel, however after the install it didn't make an emulated Xbox controller appear like I expected to show up.

If I run ScpMonitor.exe and right click on the Tray Icon and go to "XInput Tester" it appears that the controller inputs are being registered correctly, mapped to the proper buttons as well.

Here's a screenshot of all of appropriate windows:
http://zeekdageek.tv/ss/6UGhr.png

I've attempted to run ScpCleanWipe.exe and reinstall a freshly downloaded version and I'm not sure what I'm missing.

I'm not sure what I'm missing in order to make things work in games like Elite Dangerous, Binding of Isaac, Borderlands, Mercenary Kings, etc. All of which I've tested and not had the controller work properly.

Dorque, 2016-01-14:
What does it show in Device Manager? I had this problem a few versions back when my DS3 kept installing as a libusbK (spelling?) device and not the Xbox 360 Controller device.

ZeekDaGeek, 2016-01-14:
In the libusbK category of the device manager there is a Xbox 360 Controller installed there. I tried disabling it and reenabling and nothing seems to happen with it. with no effect.

I also attempted to disable it and the XInput Tester from ScpMonitor.exe seemed to still register button pressions.

Dorque, 2016-01-14:
Try right click properties on that and, under the drivers tab, click the roll back driver button

ZeekDaGeek, 2016-01-14:
The rollback button is disabled unfortunately, not something I can do.

Dorque, 2016-01-14:
I'll let you know later once I'm back on my laptop. I'm suspecting you need to install the Xbox 360 controller drivers.

Dorque, 2016-01-14:
If you have Windows 7 or earlier, did you try installing the drivers. Afterwards select Update Drivers and change it to the Xbox 360 Controller driver?

Steps
Update Driver...
Browse my computer for driver software
Let me pick from a list of device drivers on my computer

ZeekDaGeek, 2016-01-15:
Thank you Dorque, it appears that fixed the issue. <:

Here's a summary of the solution if anyone needs it in the future.

PROBLEM SUMMARY:

After installing SCPToolkit everything appears to work correctly. When you connect your PS3 Controller through wired or wireless means you get a proper connection reported by ScpTrayApp.exe or ScpMonitor.exe. Your controller also will appear to work if you run ScpMonitor.exe and select "XInput Tester".

The real problem seems to be with Xbox Controller emulation, resulting in the controller not appear in the Windows Game Controllers panel.

SOLUTION SUMMARY:

You may have an outdated / broken controller emulator installed from a previously used software version.

- Disconnect your controller
- Open Device Manager
- Go to the libusbK category and expand it.
- Right click on "Xbox 360 Controller" in the category and select "Update Driver Software..."
- Select "Browse my computer for driver software"
- Tell it to search for driver software in the location of your ScpToolkit folder (C:\Program Files\Nefarius Software Solutions\ by default)
- Ensure that "Include subfolders" is enabled and click next.
- Attempt to connect your controller again and ensure that it shows up in the Game Controller panel and you're done!

mirhl, 2016-01-16:
That should be already handled by scp installer... 
Which special VID your xbox360 device has?

ZeekDaGeek, 2016-01-16:
I'm not exactly sure. During the install process when asked I chose the only USB Input Device that existed in the DS3 drop down. There was nothing that looked like a more suitable name in the whole list and I'm not sure how to get the VID.

mirhl, 2016-01-16:
> I chose the only USB Input Device that existed in the DS3 drop down

This isn't something a person with a normal `PLAYSTATION(R)3 Controller` would say. 

> how to get the VID

You should have checked the x360 controller in windows device manager when you had the problem

ZeekDaGeek, 2016-01-17:
I'm assuming you mean when going through the ScpToolkit install process it says to select a DS3 device. I looked through the list of devices in that list and none of them said anything Playstation related so I assumed it was just the USB Input Device.

Also you say that like like I'm trying to pull one over on the Scp team by slipping by a non standard controller or something. ScpToolkit is supposed to work with a bunch of different controllers how would I benefit from saying I have on when I don't. It just didn't show in the list if that's what it's supposed to display as.

Also I should have totally gone through a list of hundreds of devices looking for a VID number that I didn't know exist to magically fix my problem that I had to submit an issue report to get help with, I have no idea why I would even want to see a VID number and what it would help with.

mirhl, 2016-01-17:
> Also you say that like like I'm trying to pull one over on the Scp team by slipping by a non standard controller or something. 

I'm saying said original controller as you said could have been using some weird driver in the first place, if it didn't show as expected. 

> Also I should have totally gone through a list of hundreds of devices looking for a VID number

Actually, you should just had to open windows device manager and look under "Human Interface Devices", but whatever. 

If it's fixed and we have no clue what could have been, I guess it can be closed.

Jan200101, 2016-01-20:
close please

closed 2016-01-20 by ZeekDaGeek