nefarius/ScpToolkit #159: Service throws exception after starting, then stops

fdavison, 2016-01-14:
After installation, the service plays the startup sound then stops.  The logs (attached) show an exception.  If I unplug the controller, start the service, it will run until I plug in the controller.

I usually install only the service and the DS3 driver.  The DS4 and Bluetooth checkboxes are not selected.  The controller is a QuadStick, which uses a USB report similar to the Afterglow controller.
[ScpDsxlogs.zip](https://github.com/nefarius/ScpToolkit/files/89839/ScpDsxlogs.zip)

nefarius, 2016-02-06:
I think there's still an issue in the code when the quadstick gets queried for its Bluetooth address - which it doesn't have. I'll look into this.

fdavison, 2016-02-14:
I'll be ready to test it when you give the word.

closed 2018-01-26 by nefarius