nefarius/ScpToolkit #160: Controller won't connect anymore after a few days after installation

Obyboby, 2016-01-14:
I own a genuine DS3 joypad and I am going crazy with trying to make it work permanently on my PC! After a while it stops being recognized, usually it's the Bluetooth "mode" that stops working, now even USB won't work. Really annoying - and that's the main reason why I tried your software! Looks clean and all, and looked like it fixed this issue I had - but unfortunately it's back. My dongle was included in my motherboard and it's an Intel AC 7260 WiFi/Bluetooth adapter. Log attached as requested.
[ScpToolkit_Log-Package_rfont_1452800840.zip](https://github.com/nefarius/ScpToolkit/files/90910/ScpToolkit_Log-Package_rfont_1452800840.zip)

mirhl, 2016-01-23:
You might want to check if DS3service is still running.

Obyboby, 2016-01-23:
Started working again, even after rebooting, not sure what fixed it.

closed 2016-01-23 by Obyboby