nefarius/ScpToolkit #161: ScpUpdater popup interruptions

DorCoMaNdO, 2016-01-15:
I find it really annoying, it's great that there's an automatic updater, but an option to disable it and/or disable the visibility of it (updates in the background) would be great, I updated to this toolkit from the original SCP 2 days ago, the popup appears once a day, yet interrupted me in both occasions.

mirhl, 2016-01-16:
Open `services.msc`, set *updater* to manual start.

DorCoMaNdO, 2016-01-16:
It seems to be on manual by default, it still runs.

mirhl, 2016-01-16:
Ok, I found the thing. 
Open windows task scheduler, and adjust *updater* activation.

nefarius, 2016-01-16:
I will refine the updater to remain silent if there are no new versions in the next release.
(I got annoyed by it too =P)

closed 2016-04-10 by nefarius

bilgekan, 2016-06-01:
That updater pop-up everytime on startup. I really get annoyed. How can i fix it? I don't know how to use that github thing, i am too oldschool for such things. So, if there is already a solution somewhere in here, sorry about it. I couldn't find it.

Btw, i already checked services.msc and it was manual.

Ty in advance

mirh, 2016-06-01:
You know, it's not like you are obliged to stop reading at second post. 

Nevertheless, update to last version.

bilgekan, 2016-06-01:
I am already using last version.

mirh, 2016-06-01:
1.7.277.16103 ?

szyszak, 2016-06-13:
I have the same problem, auto update notifications pop up almost every time I start my computer. Can't find scp anywhere in task scheduler.

jkarlsso, 2016-06-21:
So, I also have this problem... but I admittedly am not running the latest version (I have 1.6 something). But why is the updater saying the software is up-to-date when it is not? What is the use?

My questions is: How do I manually update the "correct" way? I ask because on my previous windows installation I got scptoolkit to work once and when I tried to reinstall I could not get it to work (trying everything for 3 whole days). Now I just installed windows fresh, installed scptoolkit once and it works... so now I really don't want to risk breaking it. So, how do I update with minimal risk of breaking functionality, or can I turn off the autoupdater some other way?

mirh, 2016-06-21:
Prolly because new version is supposed to be a beta. 
Or fucks knows. 

@nefarius we are all hanging by your magic fingers

nefarius, 2016-06-23:
Trust me if I currently could I would devote more time to the project =/

CoBrA2168, 2016-06-30:
I came here because I too was annoyed with the notifications...according to the log above, it looks like it's been fixed in a 1.7 version of the toolkit.  I'm running the latest official release 1.6.x, so I'm guessing I'll get this feature when 1.7 is officially released.

m4tic, 2016-07-17:
This is a scheduled task that runs at 10am everyday. You can disable it by going to task scheduler >go to task scheduler library > right click on "updater" (it has scpupdater.exe in the action tab) and disable.

nefarius, 2016-07-17:
@CoBrA2168 right.