nefarius/ScpToolkit #163: Random stuck inputs

devilxking, 2016-01-16:
Using a real Dualshock 3 controller and the latest version of Scp, running the latest version of Windows 10

It seems to install correctly but during gameplay in (just as an example) Street Fighter 4, I randomly get a constant up-left input from either the d-pad or left analog. It stops after a while but starts again randomly.

Hoshiyuu, 2016-01-16:
There is a chance that it is related to issue #126, Nefarius updated the toolkit with an option, details here: https://github.com/nefarius/ScpToolkit/issues/126#issuecomment-170225031

Tilded, 2016-01-17:
I had this issue with both a wired Dualshock 3 controller and a Hori RealArcadePro fightstick several weeks ago on Windows 7.

I recently purchased a new laptop with Windows 10 on it and I'm having the same issue.

Changing the sync/async processing setting didn't help with anything.

mirh, 2016-04-29:
I wonder if it couldn't have something to do with https://github.com/torvalds/linux/commit/8f5f0bc2766af7785161a4440bca1d0746eb3671

closed 2018-01-26 by nefarius