nefarius/ScpToolkit #165: DualShock 3 Left Analog moving on it's own every 5 minutes or so

warmbaguettes, 2016-01-17:
Exactly what I said in the title.
I'm using Windows 10 x64 (Enterprise) and the latest version of ScpToolkit with a Sony DualShock 3 controller via USB.

I've also tried disabling rumble and both the sync/async settings in the Manager but it happens the same.

Thanks in advance!

warmbaguettes, 2016-01-18:
Solved it by using a really old version (ScpServer_v1.5.5724.38523)

r4tzt4r, 2016-01-24:
Hi, how did you install ScpServer_v1.5.5724.38523 and where did you get it? I tried the one on the releases site but my DS3 does not work when I install it. The only way to make my DS3 work is when I install the drivers after installing the Scptoolkit, but then I get the same problem you reported. Hope you can help me!

warmbaguettes, 2016-01-24:
To be honest I did just that, but I had installed the old Scp drivers by the original author. Try that.

Hope nefarius looks into this issue since it seems there are quite a bunch of people experiencing it.

r4tzt4r, 2016-01-24:
So, just creating the "C:/Program Files/Starlet.Crush Production" folder, placing everything there and running the scpdrivers should work? I try that but nothing gets installed, the controllers does not get recognised. Scptoolkit actually install the drivers and it works.

warmbaguettes, 2016-01-24:
Maybe the service isn't running automatically. This happened to me. Try running Scpserver.exe and see if it works.

r4tzt4r, 2016-01-24:
Every other step I mentioned is alright? Thanks for the tip, I will try it later tonight. Thanks!

warmbaguettes, 2016-01-24:
Yes everything else should be okay. Hope it works for you.

ElephantGent, 2016-01-24:
I have the same issue. It never happened when I used the original xinput wrapper, but that old driver freezes my pc so using it is out of the question.

warmbaguettes, 2016-01-24:
Yeah it's pretty troublesome to get it running on Windows 10 also. Hope the dev sorts it out.

mirhl, 2016-01-25:
Duplicate of #163 ?

r4tzt4r, 2016-01-25:
So, to solve this I had to go back to ScpServer v1.5.5724.38523. First I uninstalled any other drivers I had, Including those of the  DS3controller. So when I plugged in it was like plugging for the first time. Then I downloaded the older Scarlet.Crush 1.2.0.160 Input Wrapper, then I put the files on the folder "C:/Program Files (x86)/Scarlet.Crush Productions", which it has to be created. Without installing anything, I added to that same folder the files of v1.5.5724.38523, replacing and adding some other files. 

Now, before running SCPDriver.exe, you have to run SCPserver.exe. After opening SCPServer you can install the drivers like you normally would with SCPDriver.exe. This worked for me. And remember that anytime you want to use the controller you have to open (and keep in the background) SCPServer.

mirhl, 2016-01-25:
Could you try to further dissect the problem?
In particular, does the problem start to happen already with following release(1.5.5730.24251)?
Or was it introduced with 1.6 or what else?

r4tzt4r, 2016-01-25:
At least in my case, I wasn't able to run any 1.6 release without facing this issue. I went back to 1.5.5724.38523 because of warmbaguettes recommendation.

warmbaguettes, 2016-01-25:
Same, any of those that include the new installer and tray notifier work the same no matter what settings.

Mind you I've only tried with a USB cable since I don't have a Bluetooth dongle right now.

mirhl, 2016-01-25:
But can you confirm the problem happeared in the next version, 1.5.5730.24251 ?

warmbaguettes, 2016-01-26:
Probably not, basically what I did was: Download latest release, install, test, problem, download previous release, install, test, same problem.

Decide to download something below 1.6 so I got that and it worked so I leaved it at that.

nefarius, 2016-02-06:
I'll prioritize this although I honestly have no idea why this bug was introduced...

closed 2018-01-26 by nefarius