nefarius/ScpToolkit #168: WinSCP Non-Dedicated Bluetooth Adapter

Teutonic84, 2016-01-18:
Hi, I'm sorry if this has been asked before, but I didn't see any posts about it. I have tried using your ScpToolkit and can't get past the fact that my laptop's bluetooth has to be dedicated for my DS3 or DS4 controller. I was curious if it was possible to implement the method that DS4Tool uses for bluetooth? This does not require a bluetooth adapter to be dedicated, so i can still use my non-DS3/DS4 devices. The only reason why i'm asking this as opposed to just using the DS4Tool, is that it does not work with DS3 controllers where your utility does. Hoping this is possible.

Thanks,

Dorque, 2016-01-18:
I don't think it's possible with the DS3 because of the way the DS3 works. The DS4 is able to connect to regular bluetooth adapters.

nefarius, 2016-01-19:
Nope; the DS4 works differently because it has a dedicated "PC-Mode" which lets it pair with other devices than the PlayStation. The DS3 doesn't have this feature so the complete PlayStation Bluetooth-Stack has to be emulated. There's no way around this.

closed 2016-01-19 by nefarius

Teutonic84, 2016-01-19:
Ah ok well that's a shame. Any chance there's a possibility for you to add the "PC mode" just for the DS4 and emulate it only for the DS3? You're program has some nice bug fixes and optimizations, but for those of us that just have a DS4 and a laptop with bluetooth, it would be nice to not have to buy a bluetooth dongle to use just for the controllers.