nefarius/ScpToolkit #169: Controller won't work anymore "Couldn't connect to server, please check if the service is running!"

Bixbies, 2016-01-19:
Alright so I was using my DS3 controller all fine and dandy last night. I plug it in today and received the error "Couldn't connect to server, please check if the service is running!". I fixed that issue by changing to delayed start up process. 

SCP recognizes my controller is plugged in and charging but the controller isn't working with any of my  games. The monitor doesn't have pop up notifications anymore telling me that the controller is unplugged or plugged in. I also reinstalled (forced) the drivers but nothing seems to work. 

Help! (please and thanks)

[ScpToolkit_Log-Package_Shanyne_1453164080.zip](https://github.com/nefarius/ScpToolkit/files/95180/ScpToolkit_Log-Package_Shanyne_1453164080.zip)

Jan200101, 2016-01-20:
chech if the service is running 
I would like to help but i am on my Phone right now

Bixbies, 2016-01-21:
I double and triple-checked to make sure it was running each time and it was.

closed 2018-01-26 by nefarius