nefarius/ScpToolkit #173: Wireless Controller not found in Driver installer. It is connected via bluetooth.

Ichigo07, 2016-01-20:
[ScpToolkit_Log-Package_Ichigo_1453294942.zip](https://github.com/nefarius/ScpToolkit/files/97456/ScpToolkit_Log-Package_Ichigo_1453294942.zip)

Jan200101, 2016-01-20:
Wireless controllers wont be picked up my the installer. You have to
connect it over cable

Sentry

Ichigo07, 2016-01-20:
I was playing with this exact setup with the old ScpServer which was working until like 2 month ago.

When i try to connect it via cable it works with the old scpServer. 

Somehow there are always 2 Xbox Controller and 1 wireless Controller showed in the Controller menu with the old scpServer.

I recognized that the controller seems to work as second controller with the old scp server.

It is an Original Dualshock 4.

Jan200101, 2016-01-20:
Do you have any virtual controllers (e.g. Vjoy) installed?

Ichigo07, 2016-01-20:
[ScpToolkit_Log-Package_Ichigo_1453295572.zip](https://github.com/nefarius/ScpToolkit/files/97467/ScpToolkit_Log-Package_Ichigo_1453295572.zip)

i installed it now via cable and its flashing blue now. But its not working ingame.

i havent installed Vjoy, i only had Scpserver (the old one) which worked 3 days ago for some hours, then i had ds4windows which i deinstalled (even the drivers) and i got this one.

Ichigo07, 2016-01-20:
[image]
[image]
[image]

Ichigo07, 2016-01-20:
this was a wired installation, if its only connected via bluetooth there is no wireless controller listed

Ichigo07, 2016-01-20:
Input tester is working tho...

Jan200101, 2016-01-20:
okay can you make a log with the controller connected over cable?

Ichigo07, 2016-01-20:
[ScpToolkit_Log-Package_Ichigo_1453296731.zip](https://github.com/nefarius/ScpToolkit/files/97499/ScpToolkit_Log-Package_Ichigo_1453296731.zip)

I really appriciate your help, i was searching for the problem 2 days now and i wasnt able to find any clue why its acting like that. because it was working with the old scpserver sometimes, even wireless.

Jan200101, 2016-01-20:
btw is the service on?

Ichigo07, 2016-01-20:
yes its active atm

Jan200101, 2016-01-20:
Okay try disableing the one that is not there

Jan200101, 2016-01-20:
its called Xbox 360 Controller for Windows

Ichigo07, 2016-01-20:
[image]

Ichigo07, 2016-01-20:
There is only one atm

Jan200101, 2016-01-20:
okay win10 is very confusing :I
i am going to replicate it and find out how to fix

Jan200101, 2016-01-20:
The Usb file in you log says that there are 2

Jan200101, 2016-01-20:
Her
1118;654;Controller (XBOX 360 For Windows);USB\VID_045E&PID_028E&IG_03\2&9C6DEB5&B&03;USB\VID_045E&PID_028E&IG_03;HidUsb;Unknown

and here 

1118;654;Xbox 360 Controller for Windows;USB\VID_045E&PID_028E\0000001;USB\VID_045E&PID_028E&REV_0114;xusb22;Unknown

Ichigo07, 2016-01-20:
On the old scpserver it was acting like that, on some games the controller worked wireless but it was always connected as second player, so i couldnt play. But the game recognized the controller.

Ichigo07, 2016-01-20:
let me try to restart then i will redo the debug lets see. brb

Ichigo07, 2016-01-20:
[ScpToolkit_Log-Package_Ichigo_1453298164.zip](https://github.com/nefarius/ScpToolkit/files/97517/ScpToolkit_Log-Package_Ichigo_1453298164.zip)

Ichigo07, 2016-01-20:
okay now it seems to work wired. so how will i be able to get it wireless working?

Ichigo07, 2016-01-20:
I tried to connect via bluetooth now by just pressing the playstation button and it said dualshock 4 controller found but the controller instantly shuts off.

Uhsheen, 2016-01-21:
I just picked up a Bluetooth adapter today and I am having the same issue.

Ichigo07, 2016-01-21:
I havent found a solution for the Bluetooth way yet.

war1ox, 2016-01-22:
I have the same connection problem, I install everything correctly and even restart. The Bluetooth is connected but wont work in any games. The version that was released within the last 2 updates worked flawlessly, and now it seems like it doesn't work at all and I have the real dualshock 3.

nefarius, 2016-02-06:
I just love it when older releases work better than newer ones -_- I will investigate...

KinanEldari, 2016-05-22:
"I tried to connect via bluetooth now by just pressing the playstation button and it said dualshock 4 controller found but the controller instantly shuts off. "

I fixed this by first doing the clean uninstall utility (if your Pc can handle it), then connecting the DS4 through Windows' regular bluetooth utility first and making sure it's paired there. After that, I ran the installation and initialization setup again and my controller once again worked. Mind, this is with the 1.7Beta drivers, but it's worth a shot, right?

closed 2018-01-26 by nefarius