nefarius/ScpToolkit #174: Everytime I install the Driver it seemingly installs a bluetooth driver as well?

Rez91, 2016-01-22:
Hi All,

This was working so well before Windows 10 too..., it seems everytime I try to install the drivers I get these errors and then my mouse stops working...but its plugged in via USB. Anyone have any ideas on why this is?

[image]

Thanks!

nefarius, 2016-01-22:
Because you selected your Logitech G400 Optical Mouse as a DualShock 3 Controller in the driver installer and overwrote its default driver?

Rez91, 2016-01-25:
Wow, thanks for the response Nefarius! :D

I tried that and even tried installing the actual drivers for the mouse, uninstalled the drivers and only clicked on the PS3 controller this time, but it still seems to think it needs to overwrite the mouse. I reinstalled the entire SCPToolkit, but no change. Is there any way to get this back to the original settings? 

[image]

DoeEensGek, 2016-01-27:
Open scp clean wipe(start menu). unplug your mouse, then open scptoolkit driver install with your keyboard. Check with spacebar and tab the right controller, install and when finished, plug back your G400.

Rez91, 2016-01-29:
@DoeEensGek This bizzarely causes my keyboard not to work haha. It seemingly overwrites at least one input device when installing. I have a usb receiver mouse so that works as a work around but I seem to have put myself in an odd situation. I kinda wonder why you have the option to overwrite non PS3 interfaces at all.

Thanks!

DoeEensGek, 2016-01-30:
Me too... also it overwrites automaticly sometimes after restart. I used a old wired non usb keyboard for the install.

But using it longer i'm really guessing why this tool is made this way, it comes with a lot of different programs, all with there own (small) functions, i would advice to put them in one program.
Also the installer it starts with, why is it made this difficult for end users? When plugin new controllers it would work automatic I understand. For bluetooth I would say put a button from the systray program where you can start a bluetooth device as a psX receiver. This way people maybe can switch.

But as I understand nevarius is working on a more friendly version.:smile:

DoeEensGek, 2016-01-31:
Try this version:
https://github.com/nefarius/ScpToolkit/releases/tag/v1.6.133.15324

Worked for me.

closed 2018-01-26 by nefarius